dusty-nv/jetson-inference #1041: Error while running run_ssd_example.py in ssd

VeeranjaneyuluToka, 2021-05-11:
As mentioned above i am trying to run run_ssd_example.py for a single sample on Jetson xavior but i am getting the below error

  File "run_ssd_example.py", line 51, in <module>
    boxes, labels, probs = predictor.predict(image, 10, 0.4)
  File "/jetson-inference/python/training/detection/ssd/vision/ssd/predictor.py", line 38, in predict
    scores, boxes = self.net.forward(images)
  File "/jetson-inference/python/training/detection/ssd/vision/ssd/ssd.py", line 93, in forward
    locations, self.priors, self.config.center_variance, self.config.size_variance
  File "/jetson-inference/python/training/detection/ssd/vision/utils/box_utils.py", line 104, in convert_locations_to_boxes
    locations[..., :2] * center_variance * priors[..., 2:] + priors[..., :2],
RuntimeError: Expected all tensors to be on the same device, but found at least two devices, cuda:0 and cpu!

would you please let me know what could be the reason?

dusty-nv, 2021-05-11:
Hmm, strange that I haven't encountered this.   In this line of code, perhaps you can try adding `.cpu()` or `.cuda()` to each tensor in the call:

https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/vision/ssd/ssd.py#L93

dusty-nv, 2021-05-11:
Hmm ok.  Just call `locations.cpu(), self.priors.cpu()` and don't call .cpu() on `self.config.center_variance, self.config.size_variance`.  Does that work?

dusty-nv, 2021-05-11:
OK gotcha, glad you got it working.  It is strange because both the priors and locations should already been on same device (GPU), but now you got it working fine anyways :+1:

dusty-nv, 2021-05-11:
That is probably the maximum defined by the grid configuration in the SSD model.

VeeranjaneyuluToka, 2021-05-11:
Ok, but i am using the engine that i have generated using this github and implemented inference separately to simplify and make use of my fwk. I am getting 18000 scores and 12000 boxes, i do not understand really what is happening there. I hope i can use engine separately, would you mind having a look at inference implementation attached and let me know if there is any mistake?
[reg_trt_infer_v1.txt](https://github.com/dusty-nv/jetson-inference/files/6460745/reg_trt_infer_v1.txt)

dusty-nv, 2021-05-11:
The height dimension of the output should be equal to the number of classes, and the column dimension should be equal to the max number of detections:

https://github.com/dusty-nv/jetson-inference/blob/2fb798e3e4895b51ce7315826297cf321f4bd577/c/detectNet.cpp#L419

When you call `self.engine.get_binding_shape(binding)` on it, what is the size reported?