dusty-nv/jetson-inference #1042: docker not working

vucic0001, 2021-05-12:
Not sure what happened I updated from the software updater on the desktop and now the docker wont fire up.

vucic0001@steve:~$ cd jetson-inference
vucic0001@steve:~/jetson-inference$ docker/run.sh
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.5.1
[sudo] password for vucic0001: 
size of data/networks:  535209937 bytes
CONTAINER:     nvcr.io/ea-linux4tegra/jetson-inference:r32.5.1
DATA_VOLUME:   --volume /home/vucic0001/jetson-inference/data:/jetson-inference/data --volume /home/vucic0001/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/vucic0001/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/vucic0001/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/vucic0001/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
Unable to find image 'nvcr.io/ea-linux4tegra/jetson-inference:r32.5.1' locally
docker: Error response from daemon: unauthorized: authentication required.
See 'docker run --help'.

dusty-nv, 2021-05-12:
Ah, ok - you need to update your git repo with `git pull origin master`.  It will update the docker scripts to look for the correct container image.

vucic0001, 2021-05-12:
Got it.
really appreciate your prompt response 
will follow the instructions

have a great day

closed 2021-05-12 by vucic0001