dusty-nv/jetson-inference #1046: The csi camera comes out flipped over.

soyeon219, 2021-05-14:
Describe the issue
Please describe the issue

The csi camera comes out flipped over. What should I do?

What version of L4T/JetPack
L4T/JetPack version: 4.3

What version of OpenCV
OpenCV version: 3.4.0

Python Version
Python version if applicable: 2.7.1

dusty-nv, 2021-05-14:
Hi @soyeon219, can you run `video-viewer --input-flip=rotate-180` and see if that fixes it?

For more info, see here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#input-options

pjpi5227, 2021-07-21:
I tried inserting that in to the my-detection.py file and I keep getting a syntax error.  When I run from the command line it works but when I run python my-dection.py it reverts to flipped.   Im a super nube.  2 days trying to fip the image.   dedication!  Any ideas?

dusty-nv, 2021-07-21:
Hi @pjpi5227, try this:

``` python
camera = jetson.utils.videoSource("csi://0", argv=["--input-flip=rotate-180"]) 
```

pjpi5227, 2021-07-22:
Dustin,

Thank you sir for your reply.  I tried it and it still is flipped.  I also
saw the other thread on github where you mentioned creating a detectnet.py
text file.  I tried that as well and the image is still flipped.  Something
that I noticed is in all of the attempts where I get a video stream   the
gstreamer command is generated in the
jetson-inference/utils/camera/gstCamera.cpp.  In that terminal I can see
were it is flipping the image.  I attached a couple screen shots.  Can this
file be edited to not flip it.  In that same terminal it refers to
videoOptions opt; and then opt.ioType  = videoOptions::INPUT;

Thanks again for your help sir

Best Regards,

PJ

dusty-nv, 2021-07-22:
Can you put some `print()` statements in your `my-detection.py` to make sure the one you are editing is actually the one that is running?

If you are editing `jetson-inference/python/examples/my-detection.py` then run this after editing:

``` bash
cd jetson-inference/build
cmake ../
make
sudo make install
```

Or make sure you are running the one you have edited:

``` bash
cd jetson-inference/python/examples
python3 my-detection.py
```

If that doesn't worry, please provide the terminal output.

pjpi5227, 2021-07-22:
Ok it worked but not exactly like you mentioned above.

I had to edit the file

jetson-inference/utils/camera/gstCamera.cpp

I opened the file in a text editor and edited the portion that refers to
MIPI cameras needing to be flipped.  Saved

then

cd jetson-inference/build
cmake ../
make
sudo make install

cd

python my-detection.py

Good to go

pjpi5227, 2021-07-22:
Dustin,

Thank you so much.  You made my week.

Per my last email i ment to say CSI cameras not MIPI.  Here is the porton
of the code i had to edit.  Also i just realized my other attachments were
junk.  My apologies.

Thanks again!

Cheers!

PJ

closed 2023-04-27 by dusty-nv