dusty-nv/jetson-inference #1050: eval_ssd.py ERROR: ValueError: could not convert string to float: '(1)'

hisu66, 2021-05-15:
Hi Dusty, thank you for your sharing about jetson-inference. I got this issue, please help me to check it!

-Problem statement:
   I already trained as SSD model and successfully exported to ONNX format. It work good on my Jetson Nano. But when I would like to evaluate this model, I got this issue.

eval_ssd.py:68: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  all_gt_boxes[class_index][image_id] = torch.tensor(all_gt_boxes[class_index][image_id])
It took 0.07900500297546387 seconds to load the model.
process image 0
Load Image: 0.040582 seconds.
Inference time:  0.008016347885131836
Prediction: 0.067816 seconds.
process image 1
Load Image: 0.036212 seconds.
Inference time:  0.00464940071105957
Prediction: 0.067012 seconds.
process image 2
Load Image: 0.037072 seconds.
Inference time:  0.004647731781005859
Prediction: 0.072039 seconds.
process image 3
Load Image: 0.035309 seconds.
Inference time:  0.00461125373840332
Prediction: 0.067192 seconds.
process image 4
Load Image: 0.034933 seconds.
Inference time:  0.004622459411621094
Prediction: 0.069671 seconds.
process image 5
Load Image: 0.036096 seconds.
Inference time:  0.004750728607177734
Prediction: 0.071544 seconds.
process image 6
Load Image: 0.035272 seconds.
Inference time:  0.005022764205932617
Prediction: 0.064818 seconds.
process image 7
Load Image: 0.034263 seconds.
Inference time:  0.004570722579956055
Prediction: 0.060545 seconds.
process image 8
Load Image: 0.037359 seconds.
Inference time:  0.004407167434692383
Prediction: 0.058782 seconds.
process image 9
Load Image: 0.035512 seconds.
Inference time:  0.00441741943359375
Prediction: 0.058493 seconds.

Average Precision Per-class:
Traceback (most recent call last):
  File "eval_ssd.py", line 209, in <module>
    args.use_2007_metric
  File "eval_ssd.py", line 81, in compute_average_precision_per_class
    scores.append(float(t[1]))
ValueError: could not convert string to float: '(1)'

Need help!!

dusty-nv, 2021-05-16:
Hi @hisu66, are you running eval_ssd.py on a PyTorch checkpoint (.pth) or your ONNX model?   It should be run on PyTorch checkpoint

hisu66, 2021-05-16:
Hi Dusty, I used PyTorch file (.pth) with the lowest loss to evaluate. I had trained another dataset and evaluate, but it still happens with the same error. Notice that I used google colab to train and export.

dusty-nv, 2021-05-16:
Could you try `run_ssd_example.py` and see if that works?   I have not actually used `eval_ssd.py` - that came from the upstream author.

hisu66, 2021-05-16:
I haven't tried run_ssd_example before but I successfully implemented onnx file to my model on jetson nano. Moreover, I had searched and followed to many documents on google but the issue still persist. So, do you have any recommendations for this model evaluation?

dusty-nv, 2021-05-16:
Have you tried replacing `torch.tensor(all_gt_boxes[class_index][image_id])` with `all_gt_boxes[class_index][image_id].clone().detach()`

Is it possible that one or more of your classes has no groundtruth boxes in the dataset?

hisu66, 2021-05-16:
Let's me try it now! I will response the result as soon as posible! Thank you for your advice, Dusty. You are my idol <3

hisu66, 2021-05-16:
hi Dusty, I replaced it as you advice, the issue still persist and similar to the old one:

Average Precision Per-class:
Traceback (most recent call last):
  File "eval_ssd.py", line 209, in <module>
    args.use_2007_metric
  File "eval_ssd.py", line 81, in compute_average_precision_per_class
    scores.append(float(t[1]))
ValueError: could not convert string to float: '(2)'

The different is '(1)' replaced by '(2)'

dusty-nv, 2021-05-16:
Hmm unfortunately I haven't dug into this before...if you were to copy your pytorch checkpoint and dataset to your Nano, does it run there, or make same error?   Trying to see if it is related to the google colab environment.   My other thought is it is related to the dataset.

hisu66, 2021-05-16:
I will try to used jetson nano to evaluate it. By the way, my dataset is a sequence of around 450 images with 4k resolution. Do you think the resolution affects it.

hisu66, 2021-05-16:
Hi Dusty, I tried to evaluate it on jetson nano, issue still persists. I think your second thought is right!

dusty-nv, 2021-05-17:
I think it has less to do with the resolution and more to do with the annotations.  Perhaps some annotation is missing or some class doesn't have annotations.   Is it possible for you to send me your dataset, .pth checkpoint, and labels.txt file?

hisu66, 2021-05-17:
Sure Dusty, this google drive link contains my data file which is located in data/vehicle and models/vehicle for .pth file (I used .pth at epoch 90 for onnx exportation) and models/vehiclessd1.onnx

https://drive.google.com/drive/folders/1zmxAuD0x210UUynuw2De7-hb_aqNHiw5

dusty-nv, 2021-05-17:
Unfortunately google drive keeps getting stuck zipping up your dataset when I go to download it.   Can you create a .tar file of your dataset and upload it instead?   I was able to download your .pth checkpoint and labels.txt without issue

I also confirmed that eval_ssd.py is working here on my example dataset, so I think the problem is related to the dataset.

hisu66, 2021-05-17:
Hi Dusty, I zipped that file as ssd.zip on this google drive link below. Please review it.
By the way, thank you for your finding, I am starting to annotate my data again. Appreciate your work <3 

https://drive.google.com/drive/u/1/folders/1faJGsG0f7EcLHtujmg4Zc3Jv_3Y4Dsn8

dusty-nv, 2021-05-17:
Ah ok, I found that because your dataset files had spaces in the filenames, it was throwing off the eval script (which saves the results in space-separated CSV).  So instead I changed the script to use tab-separate TSV, and then it worked.

https://github.com/dusty-nv/pytorch-ssd/commit/543c887453015c14e7d8610a1483402d78d4662e

BTW here were the results on your dataset:

```
Average Precision Per-class:
Motorcycle: 0.30772282700675946
Car: 0.4618906822520599
Others: 0.20354212847493858

Average Precision Across All Classes: 0.32438521257791936
```

hisu66, 2021-05-17:
omg, you save my model, Dusty. Thank you very much. Appreciate your work until now. I will try to fix my dataset filenames in the next session. <3  Love it!

dusty-nv, 2021-05-17:
No problem, happy to help.  Yes to avoid any issue I would recommend to fix the dataset filenames / imageIDs so they don't have any spaces (perhaps underscore instead of spaces).  The code _should_ work with spaces in filenames, but sometimes there are corner cases like this one.

hisu66, 2021-05-17:
Yes Dusty, I watched previous space issues and saw your recommendations about this problem. I don't think I will face to it. But I did :)) Anyway, thank you very much. You are my teacher from the beginning until now.

closed 2023-04-27 by dusty-nv