dusty-nv/jetson-inference #1051: Params for loading pretrained basenet or checkpoints.

Lyns01, 2021-05-17:
What is the difference between the three parameters?
--base-net
--pretrained-ssd
--resume

dusty-nv, 2021-05-17:
`--base-net` is the mobilenet backbone (i.e. no SSD in it yet)

`--pretrained-ssd` is a pretrained SSD model that was previously trained with train_ssd.py.   `--resume` is essentially the same minus some hyperparameter settings.  In most cases `--pretrained-ssd` seems to work better.

Lyns01, 2021-05-19:
Thank you dusty.

closed 2021-05-19 by Lyns01

anish-faiz, 2021-06-12:
Good afternoon, I am training a custom object detection model using VGG16-SSD, instead of Mobilenet SSD for better accuracy. I have trained the model, but i could not load it. Can you please help me out to load my model and run detection.