dusty-nv/jetson-inference #1052: How to filter the output of particular segmentation class

mprabhakarece, 2021-05-17:
Hi,
I can run the segmentation code successful.
Kindly let me know how to filter the particular class say I want to display only cars.

Thanks and Regards,
Prabhakar M

dusty-nv, 2021-05-17:
The easiest way without retraining the model, would be to change the colors.txt file so that the other classes have the same color.

mprabhakarece, 2021-05-17:
Thank you for the quick response.
Yes I can able to filter output by make colour of the other objects as same. 
Is it possible to do through coding.

Regards,
Prabhakar M

dusty-nv, 2021-05-17:
I don't think so without changing the low-level post-processing code in `c/segNet.cpp`

If you are using the classID mask from segNet.Mask(), you can just ignore the classes you don't want.

deemeetree, 2021-09-27:
Yes, you can, @mprabhakarece do it programmatically. You would probably need to render the original image using some other tool and overlay the bounding box on top.

closed 2023-04-27 by dusty-nv