dusty-nv/jetson-inference #1053: Feature extraction

benjaminkelenyi, 2021-05-17:
Hello,
How can I get the "n-1" layer from detectnet?

I would like to combine the IR image with Depth from a camera. To do this I thought to extract the layer that returns my bounding-box on the IR image, then with the help of an MLP to gather the features obtained from IR with Depth.

Thanks,
Benji

dusty-nv, 2021-05-17:
I think you would need to change the pytorch_ssd model so that the "n-1" layer was also provided as an output to the model.  Then when the ONNX is exported, this layer is also marked as an output layer.  Then TensorRT should have it as an output.   It would require modification to both the PyTorch code to generate the model, and also the TensorRT code in `c/detectNet.cpp` to work with the additional output layer.

benjaminkelenyi, 2021-05-18:
Hello Dusty, 

Thank you very much for your explanations. 
I want to ask you, what do you mean by "to change the pytorch_ssd model so that the "n-1" layer was also provided as an output to the model". I managed to retrain the SSD Model, and I observed that the generated model format is ".pth". 
In the first phase, I thought that the model will be in ".xml" format, and I can directly modify the output layers. 
My question is, how can I modify the ".pth" model to get the layers? Or how should I train the model to be able to get the layers?

Thanks,
Benji

dusty-nv, 2021-05-18:
Hi Benji, it means that inside the model's code, you would need to return that layer as an additional output of the model:   https://github.com/dusty-nv/pytorch-ssd/tree/master/vision/ssd

closed 2023-04-27 by dusty-nv