dusty-nv/jetson-inference #1054: make install error:  ../python/jetson no such file

jasonlove521, 2021-05-17:


dusty-nv, 2021-05-17:
Are you running the `sudo make install` from `jetson-inference/build` directory?   Can you provide the build log from terminal?

If you continue to have issues, I recommend to use the jetson-inference docker container - it is already pre-built.

jasonlove521, 2021-05-17:
@dusty-nv 
yes,i run the sudo make install in the dir of   jeson-inference/build 
here is log:
Set runtime path of "/usr/lib/python2.7/dist-packages/jetson_utils_python.so" to ""
-- Up-to-date: /usr/lib/python2.7/dist-packages/Jetson
-- Up-to-date: /usr/lib/python2.7/dist-packages/Jetson/Utils
-- Installing: /usr/lib/python2.7/dist-packages/Jetson/Utils/__init__.py
-- Installing: /usr/lib/python2.7/dist-packages/Jetson/__init__.py
CMake Error at utils/python/bindings_python_2.7/cmake_install.cmake:89 (file):
  file INSTALL cannot find
  "/home/tx2/Desktop/jetson-inference/utils/python/bindings/../python/jetson".
Call Stack (most recent call first):
  utils/python/cmake_install.cmake:74 (include)
  utils/cmake_install.cmake:360 (include)
  cmake_install.cmake:120 (include)

Makefile:117: recipe for target 'install' failed
make: *** [install] Error 1

jasonlove521, 2021-05-17:
i also don`t understand the two lines code 
install(DIRECTORY ../python/Jetson DESTINATION ${PYTHON_BINDING_INSTALL_DIR})
install(DIRECTORY ../python/jetson DESTINATION ${PYTHON_BINDING_INSTALL_DIR})
in the jetson-inference/utils/python/bindings/CMakeLists.txt, but the jetson-inference/utils/python/python/jetson don`t exist exactly

dusty-nv, 2021-05-17:
`jetson-inference/utils/python/jetson` should exists.  Those CMake `install()` commands install the Python bindings into your Python module's dir.

jasonlove521, 2021-05-17:
@dusty-nv  I follow the Quick Reference instruction to install step by step, so what is the reason to cause the mistake?

jasonlove521, 2021-05-17:
I build the repo on jetson tx2, jetson pack4.1

dusty-nv, 2021-05-17:
Can you run the following command and let me know the output?

``` bash
ls -ll /home/tx2/Desktop/jetson-inference/utils/python/python
ls -ll /home/tx2/Desktop/jetson-inference/utils/python/python/jetson
```

jasonlove521, 2021-05-17:
@dusty-nv  I  have solved this problem. i download the repo on my pc and tansfer the  file to the tx2, i guess the transmission cause some issue. i am so sorry to disturb you, thanks for your answer.

dusty-nv, 2021-05-17:
> i download the repo on my pc and tansfer the file to the tx2, i guess the transmission cause some issue. i am so sorry to disturb you, thanks for your answer.

Ah ok, gotcha - yea, my next question was going to be if you had an external SSD mounted to your TX2 which was an NTFS filesystem (because NTFS doesn't like folders that are named the same but with different capitalization).   But that explains it that you had cloned it to an NTFS filesystem first, because it lost the lower-case folders.  Glad you got it figured out.

closed 2023-04-27 by dusty-nv