dusty-nv/jetson-inference #1057: Is possibile to train the model with an input size greater than 512x512?

alessandrolombardini, 2021-05-19:
I read a lot of issues about training the model with an input of 512x512 to detect far objects.  Does someone know if it is possible to train the model with bigger inputs? 

Probably the answer is still no, but maybe something has changed since the last time someone said that.

dusty-nv, 2021-05-19:
Hi @alessandrolombardini, I think it should be possible if you were to compute/figure out what the proper SSD box specs would be that go here:

https://github.com/dusty-nv/pytorch-ssd/blob/res-512/vision/ssd/config/mobilenetv1_ssd_config.py

Unfortunately I don't know how to do these myself, so only guessing.

alessandrolombardini, 2021-05-30:
Very useful: https://github.com/dusty-nv/jetson-inference/issues/931

closed 2021-05-30 by alessandrolombardini