dusty-nv/jetson-inference #1060: How to train semantic segmentation

mprabhakarece, 2021-05-21:
Hi,
This repo really helped me a lot. 
Is there any specific reason for not adding procedure for semantinc segmentation re-training and data collection.
I am facing problem in training semantic segmentation model.
In below link required code is available.
https://github.com/dusty-nv/pytorch-segmentation/tree/dev

I am new to python so facing problem in executing it.
It will be grateful if I get needed support.

Thanks in advance.

shireen1512, 2021-05-21:
Hi, I contacted dusty a while ago for this question and it was nice of him to answer me, he gave me this link, which explains how to do this task.   https://www.highvoltagecode.com/post/edge-ai-semantic-segmentation-on-nvidia-jetson
I hope it helps.

mprabhakarece, 2021-05-24:
@shireen1512 Thank you for your suggestion. With the help of above link I have trained my model and executed successfully.
But my entire image is marked as single class. kindly share your suggestion If any.

Thanks again.

shireen1512, 2021-05-25:
unfortunately I have not tried this myself, my dataset is not ready yet. sorry. 
maybe close this issue and open another maybe someone else could help

mprabhakarece, 2021-05-25:
@shireen1512 After some random changes It is working now. I am not sure problem with path to onnx file or engine file.
As of now I am closing this issue. Thank you.

closed 2021-05-25 by mprabhakarece