dusty-nv/jetson-inference #1061: Error while loading shared libraries in docker container

andrejlevkovitch, 2021-05-21:
Hi! I tried to run segnet-console from docker image, but I got error about loading shared libraries, like:

```
./segnet-console: error while loading shared libraries: /usr/lib/aarch64-linux-gnu/libnvinfer.so.7: file too short
```

docker image that I use: `dustynv/jetson-inference:r32.4.4`

Output of jetson_release:
```
[1m - NVIDIA Jetson Xavier NX (Developer Kit Version)(B[m
   * Jetpack 4.4.1 [L4T 32.4.4]
   * NV Power Mode: [32mMODE_10W_2CORE(B[m - Type: [32m3(B[m
   * jetson_stats.service: [32mactive(B[m
[1m - Libraries:(B[m
   * CUDA: 10.2.89
   * cuDNN: 8.0.0.180
   * TensorRT: 7.1.3.0
   * Visionworks: 1.6.0.501
   * OpenCV: 4.1.1 compiled CUDA: [31mNO(B[m
   * VPI: 0.4.4
   * Vulkan: 1.2.70
```

Why it happens?

andrejlevkovitch, 2021-05-21:
Looks like I found the problem: I added `"default-runtime": "nvidia"` in `/etc/docker/daemon.json`. But, if I understood [this doc](https://github.com/dusty-nv/jetson-containers#docker-default-runtime) rightly the setting is needed for build container. In my case I use container that already built.

andrejlevkovitch, 2021-05-21:
Looks like the setting is [required for run](https://docs.nvidia.com/dgx/nvidia-container-runtime-upgrade/index.html) too

closed 2021-05-21 by andrejlevkovitch

dusty-nv, 2021-05-21:
Hi @andrejlevkovitch, are you using the `docker/run.sh` script?   If so, it automatically uses `--runtime nvidia`