dusty-nv/jetson-inference #1063: video-viewer strange cpu usage

pbouzon-iter, 2021-05-24:
I'm on Jetson Nano 4GB (JetPack 4.5) and testing the scripts provided as examples with docker. On executing the video-viewer, I encountered the following strange results:

**Test 1**
**video-viewer** --input-codec=h265 rtsp://user:pass@192.168.0.106
**cpu usage:** ~107%

**Test 2**
**video-viewer** --input-codec=h265 --input-height=1080 --input-width=1920 rtsp://user:pass@192.168.0.106
**cpu usage:** ~23%

My camera resolution is 1920x1080 and streams 25 fps. Why there is such a difference on cpu utilization just by adding the input-width and input-height parameters? Both commands give the desired output (i'm able to see the image from the camera on my screen).

dusty-nv, 2021-05-24:
Hi @pbouzon-iter, specifying `--input-width` and `--input-height` will use the hardware rescaler to change the resolution.  Are you sure it isn't running at a different resolution without `--input-height=1080 --input-width=1920`

pbouzon-iter, 2021-05-25:
Hi @dusty-nv, i'm pretty sure it's running with the expected resolution. That's what i see on the terminal when running the video-viewer without specifying the `--input-width` and `--input-height`.

[image]

pbouzon-iter, 2021-05-25:
@dusty-nv i've made a mistake on checking the cpu usage when setting the `--input-width` and `--input-height` parameters, now i'm having both ways with ~107% cpu usage.

Thank's for your reply and sorry for taking your time.

closed 2021-05-25 by pbouzon-iter