dusty-nv/jetson-inference #1065: jetson-inference integration with ZED

kunalkumthekar, 2021-05-25:
@dusty-nv I am trying to integrate StereoLabs ZED 2 with detectnet (python interface) . I am using Jetson Xavier NX Jetpck 45. I was going through #625. However I am unable to install ZED SDK and hence ZED-Python API on to the docker container. I am a begineer with Docker and donot know if this is possible too. I would appreciate if you could provide help with the integration.

dusty-nv, 2021-05-25:
Hi @kunalkumthekar, I haven't tried installing ZED SDK in docker container before, is there a particular error you are getting when trying to install it?

If it's easier, you can just build jetson-inference from source outside of docker and forgo the container.

kunalkumthekar, 2021-05-25:
@dusty-nv the ZED SDK in order  to initiate the installation, doesn't permit the user to be root. The docker container is in root as it executes, hence the installation is not permitted. But as you mentioned, its always easier to build it in user environment from source, giving the user complete freedom to utilise the beauty of an algorithm written.

kunalkumthekar, 2021-05-26:
@dusty-nv  I built the jetson-inference outside docker and was trying to integrate zed to it by following the issue #625 . However was not able to get an output inference stream at the end. Getting the following error:
`Traceback (most recent call last):
  File "detectnet_zed.py", line 93, in <module>
    if not input.IsStreaming() or not output.IsStreaming():
AttributeError: 'builtin_function_or_method' object has no attribute 'IsStreaming'
[cuda]      invalid argument (error 1) (hex 0x01)
[cuda]      /home/kunal/jetson-inference/c/detectNet.cpp:68
[cuda]      invalid argument (error 1) (hex 0x01)
[cuda]      /home/kunal/jetson-inference/c/detectNet.cpp:76
Segmentation fault (core dumped)
`

closed 2021-05-26 by kunalkumthekar