dusty-nv/jetson-inference #1066: How did you convert .onnx file to .engine file?

priyaganaboor-hash, 2021-05-25:
Hey!  @dusty-nv 

Actually, I used the imanet.py and converted my .onnx to TensorRT and tested the working using the live camera as well as individual pictures. I understood your code on the training model, testing, and even converting it to .onnx. After this, I didn't understand how are you generating the engine file from onnx file, and processing the images from the camera to classify.

Now after executing,
$ imagenet.py --model=models/PlantClassification/resnet50.onnx --input_blob=input_0 --output_blob=output_0 --labels=data/PlantClassification/labels.txt csi://0 
The .engine files have been saved to .../classification/model/......engine
And also how to use the .engine file to perform classification?

dusty-nv, 2021-05-25:
Hi @priyaganaboor-hash, you don't need to explicitly use the TensorRT .engine file.  When you load the ONNX model, it looks for the .engine file in the same directory.  If it finds it, it loads the .engine instead of the ONNX.  If it doesn't find it, it creates the .engine from the ONNX and saves it, so the next time you run it will automatically find the .engine

priyaganaboor-hash, 2021-05-25:
Hi @dusty-nv ,

how to use the .engine file and perform classification?

priyaganaboor, 2021-05-25:
Hey @dusty-nv ,

I have framed my question wrong. I basically have to understand this below part.

"

// 
Classifyint imageNet::Classify( float* confidence )
{		// process with TRT
 	if( !Process() )	
{		LogError(LOG_TRT "imageNet::Process() failed\n");		
  return -1;	}		
PROFILER_BEGIN(PROFILER_POSTPROCESS);
	// determine the maximum class	

  int classIndex = -1;	
 float classMax = -1.0f;		
//const float valueScale = IsModelType(MODEL_ONNX) ? 0.01f : 1.0f;
	for( size_t n=0; n &lt; mOutputClasses; n++ )	
{		const float value = mOutputs[0].CPU[n] /** valueScale*/;				
          if( value &gt;= 0.01f )			
      LogVerbose("class %04zu - %f  (%s)\n", n, value, mClassDesc[n].c_str());			
   if( value &gt; classMax )		
{			classIndex = n;			
                      classMax   = value;		}	}		

if( confidence != NULL )		
*confidence = classMax;		
//printf("\nmaximum class:  #%i  (%f) (%s)\n", classIndex, classMax, mClassDesc[classIndex].c_str());
	
PROFILER_END(PROFILER_POSTPROCESS);		

return classIndex;}

"

dusty-nv, 2021-05-25:
Inside the `imageNet::Process()` function, it executes the model with TensorRT.

The rest of the `imageNet::Classify()` function is picking the argmax of the model output - the class with the highest probability.

priyaganaboor-hash, 2021-06-15:
Hello @dusty-nv,

With these probability values when there is no object in front of the camera, the classifier classes this as class A or class B.

Ideally when there is no object infront of the robot it should not classify at all ryt?? Is it problem with dataset or will have to changes the threshold values?
Can you help me its urgent?

dusty-nv, 2021-06-15:
You need a third class for 'background' that has images with no object in front of the camera

stephen-derosa, 2023-12-11:
@dusty-nv with respect to your comment:
`Hi @priyaganaboor-hash, you don't need to explicitly use the TensorRT .engine file. When you load the ONNX model, it looks for the .engine file in the same directory. If it finds it, it loads the .engine instead of the ONNX. If it doesn't find it, it creates the .engine from the ONNX and saves it, so the next time you run it will automatically find the .engine`

I am looking to do something similar. Can you please point me in the code where this is done?

dusty-nv, 2023-12-12:
> I am looking to do something similar. Can you please point me in the code where this is done?

Hi @stephen-derosa, you can find it here:

https://github.com/dusty-nv/jetson-inference/blob/129b892c7863ca00edb96f6f3fcbabc6c54965a6/c/tensorNet.cpp#L1213

I had also added a checksum for the models, so that if the onnx changes, the TRT engine will be re-generated.