dusty-nv/jetson-inference #1068: header not found when including "tensorConvert.h"

xmba15, 2021-05-25:
Hi @dusty-nv Thank you for the project.
I am using the c++ jetson-inference library by calling find_package(jetson-inference) in CMakeLists.txt
However, when I included <jetson-utils/tensorConvert.h>, it could not find cudaUtility.h (which I think belongs to jetson-utils).
I think the cause is due to the relative paths in tensorConvert.h as can be seen in here for the line 27-28.
https://github.com/dusty-nv/jetson-inference/blob/master/c/tensorConvert.h#L27-L28

which should be
```c++
 #include <jetson-utils/cudaUtility.h>
 #include <jetson-utils/imageFormat.h>
```

Can you please confirm this?

dusty-nv, 2021-05-25:
Can you also call `find_package(jetson-utils)` in your CMakeLists.txt?  Like so:

```
find_package(jetson-utils)
find_package(jetson-inference)
```

xmba15, 2021-05-25:
@dusty-nv Thank you for the quick reply.
Yes, I already include both jetson-utils and jetson-inference in CMakeLists.

```
find_package(jetson-utils REQUIRED)
find_package(jetson-inference REQUIRED)
```
<jetson-utils/cudaUtility.h>  and <jetson-utils/imageFormat.h> have been recognized with these two find_packages

xmba15, 2021-05-25:
For example this will fail

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.10)

project(test_jetson)

find_package(jetson-utils REQUIRED)
find_package(jetson-inference REQUIRED)

add_executable(test_jetson test.cpp)
target_link_libraries(test_jetson
  PUBLIC
    jetson-utils
    jetson-inference
)
```

test.cpp

```c++
#include <jetson-inference/tensorConvert.h>

int main(int argc, char *argv[])
{
  return 0;
}
```

dusty-nv, 2021-05-25:
OK, if you change tensorConvert to how you suggested, does it work?

tensorConvert.h isn't typically a public-facing API, which is why I probably have not encounter this before.

xmba15, 2021-05-25:
yes, it did work with  #include <jetson-utils/cudaUtility.h>
I notice you also this type of including in tensorNet.h
https://github.com/dusty-nv/jetson-inference/blob/master/c/tensorNet.h#L32-L36

Do you think it is okie if I push a PR to change that include style in tensorConvert.h?
Actually, I need to change a little bit to use a model in jetson-inference that is not currently supported. It would be great if users can customize this library for inference of more models.

dusty-nv, 2021-05-25:
> Do you think it is okie if I push a PR to change that include style in tensorConvert.h?
Sure, if the main project still builds with it like that, I will accept it no problem.  Thanks.

Typically support for different types of models would be done by creating a new type of class like imageNet, detectNet, and segNet do.  Then inside your class you implement the pre/post-processing required.

xmba15, 2021-05-25:
Thanks. It was yolov4 detection model actually. So I reused detectNet's structure.

xmba15, 2021-05-27:
@dusty-nv please check this when you have time.
https://github.com/dusty-nv/jetson-inference/pull/1069

dusty-nv, 2021-05-27:
Thanks @xmba15, just merged this.

closed 2021-05-28 by xmba15