dusty-nv/jetson-inference #1070: Training taking much space in memory

mprabhakarece, 2021-05-26:
Hi,
I am trying to train semantic segmentation with resnet18.
But my storage capacity goes to 0 MB due to model.pth of each epoch storing to memory.
kindly do any suggestion.
Thanks in advance.

dusty-nv, 2021-05-26:
Hi @mprabhakarece, you can have the checkpoint always saved to the same file by changing this line of code to the following:

https://github.com/dusty-nv/pytorch-segmentation/blob/9f95e8d30a6a13a17160d98b52ed920f02d84576/train.py#L299

``` python
checkpoint_path = os.path.join(args.model_dir, 'model.pth')
```

It will then always save the latest checkpoint to `model.pth` (and not store them all individually).  The best model would still be saved to `model_best.pth`

mprabhakarece, 2021-05-27:
@dusty-nv Thank you for the suggestion. I am getting below error while trying change path to model.pth and trying to resume from the same.
RuntimeError: [enforce fail at inline_container.cc:145] . PytorchStreamReader failed reading zip archive: failed finding central directory

without changing the script I tried deleting the pth files and resume, it is starting from the 0 i.e. model_0.pth
Kindly do needful.

dusty-nv, 2021-05-27:
What is the size of the model.pth that gets saved?   Can you paste the code changes that you made?

closed 2023-04-27 by dusty-nv