dusty-nv/jetson-inference #1073: body part segmentation

roga76, 2021-05-27:
Hi,
is there a Multi Human Body Part Segmentation known similar to Dense Pose on Nvidia Jetson Nano, NX or AGX?
Rolf

dusty-nv, 2021-05-27:
Hi @roga76, if you mean this https://github.com/facebookresearch/DensePose , I haven't tried it.  Although it looks like that project is now part of Detectron2.  I think some people have built that for Jetson if you look around the forums.

roga76, 2021-05-28:
Hi,
i tried it, but the performance is very bad. Your solution with MHP is much more performant, but the accuracy to distinguish every body part is for our application not good enough. Do you see any chance to to distinguish left arm/right arm also left/right legs in colors and with higher resolution on your solution? Sorry for beginner questions..

dusty-nv, 2021-05-28:
When I trained MHP model, I had to reduce the number of classes and combine some classes such as left arm / right arm to get it to fit in 21 classes.   So in theory you could give up some other classes and keep the separate arms / legs.  Unfortunately it was a few years ago I did this and I don't have the documentation.  However here is the remapping script for MHP:  https://github.com/dusty-nv/pytorch-segmentation/blob/dev/datasets/mhp_remap.py

closed 2023-04-27 by dusty-nv