dusty-nv/jetson-inference #1076: how to connect to jetson nanos together

Gianni2045, 2021-05-28:
I need to know how to connect 2 jetson nano development kits together without the carrier board
is there a special file for this? I connected them with USB and nothing happens

dusty-nv, 2021-05-28:
Probably the fastest/easiest way would be over gigabit ethernet

Gianni2045, 2021-05-29:
And will it show up 2 nanos in system resources?

Gianni2045, 2021-05-29:
Thank you very much for your fast support

Gianni2045, 2021-06-02:
Please Dusty I need your help I connected just using ethernet but it doesn't work it does not share any resources and I tried Gary explains and I only got as far as it says 2 nodes passed the example but in reality nothing is shared

Gianni2045, 2021-06-02:
Using Simple MPI

Gianni2045, 2021-06-02:
Please help as soon as possible

Gianni2045, 2021-06-02:
Your help is the only thing I have

Gianni2045, 2021-06-03:
I am asking as nicely as possible and you decide not to answer

Gianni2045, 2021-06-03:
I really appreciate your previous help but now this is unacceptable

dusty-nv, 2021-06-04:
Sorry I am away on vacation, and your question is unrelated to this repo or even the Jetson in particular.  It is the same as if you were connecting two PCs or two computers over ethernet.  MPI can be used to distribute workloads but they are still two separate computers with separate resources - for example, you can't directly control the GPU from another machine.

Gianni2045, 2021-06-04:
It was posted as jetson cluster super computer by Jerry explains, thank you for making me feel like I ask dumb questions unrelated to reality. I really appreciate your response even though on vacation I apologize for insulting you without knowing you were busy. Since it was talking specifically about connecting jetson nanos it is obvious why I would think it is related to jetson nano

Gianni2045, 2021-06-04:
Gary Explains on github is misleading

dusty-nv, 2021-06-07:
No worries @Gianni2045 - you can post general questions about Jetson Nano on the Nano forum here:   https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-nano/76