dusty-nv/jetson-inference #1085: how to convert segnet mask output to cv::Mat Opencv c++

mprabhakarece, 2021-06-02:
As discussed in [](https://github.com/dusty-nv/jetson-inference/issues/922)

I am trying to add 
cv::Mat cv_image(cv::Size(imgWidth, imgHeight), CV_8UC3, imgMask);
I am facing problem during compilation.
I think Opencv is not linked properly.
I changed BUILD_EXPERIMENTAL as YES to enable opencv usage. Correct If I am wrong.
Help me.

mprabhakarece, 2021-06-03:
I have disabled BUILD_EXPERIMENTAL and made Opencv as open from BUILD_EXPERIMENTAL. I can able to add opencv code. But I was able to run only one time. If do any changes again and build the code,  it is not getting impacted. Kindly do needful.

dusty-nv, 2021-06-07:
> If do any changes again and build the code, it is not getting impacted.

Did you re-run `sudo make install` after your code changes?

``` bash
$ cd /path/to/jetson-inference
$ cd build
$ cmake ../
$ make
$ sudo make install
```

Or if you are using the container, you need to build the container again.

mprabhakarece, 2021-06-08:
@dusty-nv Yes. After re-run install also no impact with the change. Even I tried deleting build folder and build again but no use. I will try again and let you know.

mprabhakarece, 2021-06-15:
@dusty-nv making change in /path/jetson-inference/examples/segnet and running

$ cd /path/to/jetson-inference
$ cd build
$ cmake ../
$ make
$ sudo make install

and running my custom generated engine file from /path/jetson-inference/python/training/segnet as woking directory

mprabhakarece, 2021-06-16:
@dusty-nv I have deleted the entire jetson-inference folder and started freshly and doing the same process. Now everything working fine. But I was not able to spot the exact problem. Thank you for your support. Closing now.

closed 2021-06-16 by mprabhakarece