dusty-nv/jetson-inference #1088: How to give commands to Docker through the shell using Python script

saifurrehman4114, 2021-06-04:
Hey,

I am writing a script of python which would run the docker first of the SSD model then change the directory into the SSD directory and then run my trained model for inference.

I am able to run the docker but unable to commands into through the python subprocess module.Any idea ho to solve this issue 

Thanks in Advance
 
Code I wrote till now :

import os
import subprocess
os.chdir("jetson-inference")

#os.system("docker/run.sh")
#os.chdir("python/training/detection/ssd") 

subprocess.call([
    'docker/run.sh', 'cd ./python/training/detection/ssd'
    ])

dusty-nv, 2021-06-07:
Hi @saifurrehman4114, try using the `-r` option to `docker/run.sh` for the command that you want to run:

https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/docker/run.sh#L34

``` python
subprocess.call([
'docker/run.sh -r detectnet.py --model=python/training/detection/ssd/models/my_model.onnx'
], shell=True)
```

saifurrehman4114, 2021-06-08:
Thanks  @dusty-nv   it worked!!!

closed 2021-06-08 by saifurrehman4114