dusty-nv/jetson-inference #1090: Detection of objects after semantic segmentation

juanpablo-sanchez, 2021-06-07:
Hello
First of all, thank you very much for the work you do in developing this packages. I am working in a project in which I have a few hundreds images annotated using irregular polynomials around the areas of interest. With this dataset I have managed to do pretty decent segmentation using the package by @Onixaz (https://github.com/Onixaz/pytorch-segmentation) considering a ResNet18. My objective is to detect the position of the different objects i have segmented within each frame. Thus the question is the following:
Is there are any implementation to conduct an object detection from the output of the functions in, for example, segnet.py: 
https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/segnet.py
I mean, taking buffers object as input and getting the number of regions (groups of pixels) in which a given class is counted.
Other option could be training some model to detect objects, but this must be done using the images I have which, as I said, are annotated with irregular polynomials.
Any suggestion, idea or comment is more than welcome.
Thank you
Juan Pablo
P.D.: Attached you can see the segmentation i got so far. As you can see I want to detect each rabbit individually.

https://user-images.githubusercontent.com/32646070/121073497-e1c86b80-c7d2-11eb-945c-acb480d7b60e.mp4

dusty-nv, 2021-06-07:
Hi @juanpablo-sanchez, I believe you mean blobbing or clustering or connected component.  I haven't done it with segnet, but I think you can do it with OpenCV or scipy:   https://stackoverflow.com/a/46738834

dusty-nv, 2021-06-07:
BTW since the rabbits are often touching, those touching would get counted as one blob...

juanpablo-sanchez, 2021-06-07:
Thank you Dustin
I was using Mask R-CNN on cpus: good results but slow (see attachment) tried to move the trained model to my jetson nano, but not success (yet). So i decided to move to your jetson-inferences.
I believe the solution based on clustering you propose it is not feasible, animals touch to each other very frequently... I am thinking on running both object detection (detectnet) and segmentation (segnet). Thus, I think we could better split the individuals in the mask after the segmentation.
In order to train the detection, can i use the same json files (with the format generated by Labelme) that i have used for training the segmentation? I followed this tutorial:
https://www.highvoltagecode.com/post/edge-ai-semantic-segmentation-on-nvidia-jetson
Or alternatively, should I modify the annotation to define squared boxes around each individual rabbit
Thank you again 
Juan Pablo

https://user-images.githubusercontent.com/32646070/121085995-e85edf00-c7e2-11eb-89af-c481fc09cbab.mp4

wukangqian, 2021-07-31:
Hi Juan,
I did a similar task (nail detection from hands + nail seg). I believe it's easier to create the detection label from seg label.
Thanks for the detailed description of your issue.
Regards,
Kangqian

closed 2023-04-27 by dusty-nv