dusty-nv/jetson-inference #1092: imagenet with trained model

nikk-cam, 2021-06-08:
Hello,
First time user. I was able to get through the training example with my 1st trained model based on [https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md](url).

I have written a script but cannot figure out how to get it execute. Perhaps this means I have to only use images and not video?

```
NET=models/tst2
 DATASET=data/tst2
 VIDEOPATH=/somepath
 
 imagenet.py --model=$NET/resnet18.onnx input_URI=$VIDEOPATH/00123.mp4  output_URI=$VIDEOPATH/00123.mp4  --labels=$DATASET/labels.txt --network=resnet18
```
Here is the error that I am getting which suggests I am not in the right ballpark!
```
[TRT]    device GPU, /usr/local/bin/networks/ResNet-18/ResNet-18.caffemodel initialized.
[TRT]    imageNet -- loaded 1000 class info entries
[TRT]    imageNet -- networks/ResNet-18/ResNet-18.caffemodel initialized.
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- couldn't find file 'input_URI=file///somepath/00123.mp4'
[gstreamer] gstDecoder -- failed to create decoder for file:///home/user/tests/jetson-inference/python/training/classification/input_URI=file///somepath/00123.mp4
Traceback (most recent call last):
  File "/home/user/tests/jetson-inference/python/examples/imagenet.py", line 58, in <module>
    input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device
```
I have verified that the input path and videos are correct. I played back the mp4 file on NX using ffplay.

I was planning to write a python script that would use the input video to capture frames, then feed the frames into inference and get classification info, write an overlay with the class info on to image, and write to an output file.
Upon inspection of imagenet.py I came to the conclusion that it was doing exactly that!

Any suggestions would be much appreciated.
Thanks,
Nikk

dusty-nv, 2021-06-08:
> `input_URI=$VIDEOPATH/00123.mp4  output_URI=$VIDEOPATH/00123.mp4`

Hi @nikk-cam, it looks like you are missing the `--` in front of `--input_URI` and `--output_URI`

Or you should be able to run it like this using positional args:

```
imagenet.py --model=$NET/resnet18.onnx --labels=$DATASET/labels.txt --network=resnet18 $VIDEOPATH/00123.mp4 $VIDEOPATH/00123.mp4
```

Note that this will overwrite your video since the input/output are the same - you probably want to specify a different output filename.

nikk-cam, 2021-06-08:
Hi @dusty-nv,
I had tried --input_URI before I opened this ticket. I do not have a copy of how I tried it. But your correction worked.
Thank you for the prompt response.

closed 2021-06-08 by nikk-cam