dusty-nv/jetson-inference #1093: Number of image in current image batch from the epoch incorrect

hantermatty, 2021-06-08:
I produced my own set of data to be able to train, creating two classes (silos, chimneys) and instead of using the camera capture tool I inserted the images directly into the desired folders, also creating the txt label file. The photos included are 100 of silos and 100 of chimneys in the train folder and 10% in val, while in test none. When I go to train the total number of images in the current batch is 28. How is this possible?

[image]

hantermatty, 2021-06-08:
Currently I have a json and csv file exported from LabelBox with inside them the image segmentation info of these images. Could it be used as a model, converting it to an onnx file (I have no idea, it's just a thought I'm doing) ???

hantermatty, 2021-06-08:
I tried to reread this sentence more carefully:

Multiply the numbers in brackets by the batch size (e.g. batch [100/625] -> image [800/5000])

Maybe this is the answer?

If so, multiplying the number 28 by 8 I should find myself with the total number of images I have in the two folders of the two classes ... I feel so stupid XD

dusty-nv, 2021-06-08:
> Multiply the numbers in brackets by the batch size (e.g. batch [100/625] -> image [800/5000])
> 
> Maybe this is the answer?

Yes this is correct, those counters in the terminal output show the number of batches.  So multiplying by the batch size, you should arrive at the number of image.s

hantermatty, 2021-06-08:
> > Multiply the numbers in brackets by the batch size (e.g. batch [100/625] -> image [800/5000])
> > Maybe this is the answer?
> 
> Yes this is correct, those counters in the terminal output show the number of batches. So multiplying by the batch size, you should arrive at the number of image.s

thanks for answering me. what about the model how can i do? I would like to be able to create my own model with its weights and to be able to use it for training. with the current models I can't get the expected good result.

dusty-nv, 2021-06-08:
From your screenshot, I see it is at 75% accuracy, which is not too bad to start.  I think you would need to add more images to dataset to get better accuracy.

hantermatty, 2021-06-08:
> From your screenshot, I see it is at 75% accuracy, which is not too bad to start. I think you would need to add more images to dataset to get better accuracy.

ok ok, so you are advising me to focus on improving this training instead of thinking about a new network made from scratch ... right?

hantermatty, 2021-06-08:
ok. currently, when I start the camera, I only notice the accuracy percentage at the top left. If I also want to see the boxes, having already trained the network, what can I do?

hantermatty, 2021-06-08:
if you like, when I present my thesis I can quote you by sharing this short discussion. besides being the author you are also of great help to me and I am very pleased to be able to share this moment.

hantermatty, 2021-06-08:
I have one last question. in case I wanted to switch to detecion mode, would there be a way to be able to label a set of images in a folder instead of taking the photo? it would be the final stage of my training and I would like to be able to view the colored mask to the specific class as well but I already have the images downloaded.

hantermatty, 2021-06-11:
Hello. I just finished the label part of the images on cvat, as you advised me and exported everything in a PASCAL VOC file. I extracted the contents in ssd / data inside a "refineries" folder and started the training, but it gives me this error on the missing model.

[image]

to remedy the error I tried to insert the previous results obtained in classification into the "models" folder, but I did not obtain positive results.

[image]

How can I get out of this stalemate by not having this mobilenet model with me? Do I have to download it and put it in the "models" folder?

hantermatty, 2021-06-11:
I edited the last comment to not be too verbose. I downloaded the missing model and started training and this is the error message I am currently getting. 
[image]

dusty-nv, 2021-06-11:
Hi @hantermatty, it looks like your labels.txt is missing `chimneys`

hantermatty, 2021-06-12:
the content was exported with its label file, called labelmap. I tried to rename it, as it had both classes in it, but nothing has changed. However I am finding in the xml files that in the export from cvat to pascal voc only the shape rectangles are considered and not the polygon shapes. I spent a lot of time carefully delineating many objects and now they have not been considered in the xml files. Am I bound to have to insert only rectangles on cvat or can I find another way with polygons, since I've already outlined each object?

dusty-nv, 2021-06-14:
Probably the easier way would be to write a tool that converted the polygons to bounding boxes

closed 2023-04-27 by dusty-nv