dusty-nv/jetson-inference #1094: Crash on running the train_ssd.py

gitgkk, 2021-06-08:
Hi Dustin,
I followed your tutorial for "Training the SSD-Mobilenet Model"
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md

I used the https://www.makesense.ai/ to annotate the images and provide the data. 

I followed the video tutorial for configuring directories and content for retraining the ssd model. When I execute the 
python3 train_ssd.py --data=data/object --model-dir=models/object --batch-size=2 --epochs=30

I get illegal instruction (core dumped) error. Impossible to know or debug what is going on. Can you please help?

Thanks !
Kashyap

dusty-nv, 2021-06-08:
Hi @gitgkk, are you able to run the following command `python3 -c 'import numpy'`?   Does it give segfault?

If so, try running `export OPENBLAS_CORETYPE=ARMV8` in your terminal first.  For more info, see here:  https://github.com/numpy/numpy/issues/18131#issuecomment-756140369

gitgkk, 2021-06-09:
Hi @dusty-nv 

Thanks for the response. That solved the problem.

Now when I run the same train_ssd script. I get the error ModuleNotFoundError: No modle named 'torchvision'.

gitgkk, 2021-06-09:
Hi @dusty-nv 
Installed torchvision. Now when I run the script, I get this
Traceback (most recent call last):
  File "./train_ssd.py", line 22, in <module>
    from vision.datasets.open_images import OpenImagesDataset
  File "/home/leap/workspace/jetson-inference/python/training/detection/ssd/vision/datasets/open_images.py", line 4, in <module>
    import pandas as pd
  File "/usr/lib/python3/dist-packages/pandas/__init__.py", line 58, in <module>
    from pandas.io.api import *
  File "/usr/lib/python3/dist-packages/pandas/io/api.py", line 19, in <module>
    from pandas.io.packers import read_msgpack, to_msgpack
  File "/usr/lib/python3/dist-packages/pandas/io/packers.py", line 68, in <module>
    from pandas.util._move import (
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xfd in position 0: invalid start byte

dusty-nv, 2021-06-09:
Hmm...can you try running the jetson-inference docker container and see if you get same error?

gitgkk, 2021-06-11:
Hi @dusty-nv 
I tried the docker. It works pretty well.
My only disappointment with docker is its high overhead for memory and cpu. My view is that the native installation should work flawlessly. Embedded devices can't afford to run docker for deploying applications. It doesn't come cheap.

Secondly, when Nvidia is taking the pain to prepare a jetpack, the other environment support for tensorflow or pytorch should also work without issues. The python and all other library versions should be configured at some place from where an install script can pick up.

While doing the native installation I ran into so many issues, the installation process is broken at many places.  Incompatibilities wasted almost 3-4 days of my time.

I would love to see a process where as hardware manufacturer Nvidia provides a full support in an ecosystem to cleanly configure the AI/ML environment.

Thanks for your help.

Kind regards,
Kashyap

dusty-nv, 2021-06-11:
Docker runs natively and is not a virtual machine - it should not have overhead.  I get the same performance inside of docker as outside. It is the preferred mechanism for deploying complex ML applications.

closed 2021-06-19 by gitgkk