dusty-nv/jetson-inference #1095: Unable to use Pyttsx3 in Docker

saifurrehman4114, 2021-06-09:
HI @dusty-nv I have installed pyttsx3 then in the my_project folder I made a python name in which I wrote a code related to that it works properly.

Code:

import pyttsx3
engine = pyttsx3.init()
engine.say(“I will speak this text”)
engine.runAndWait()

But this error show up.

[image]

And also in the detect net file https://github.com/dusty-nv/jetson-inference/blob/627b5890b49449573b7c1af8de22ce985fc395e4/python/examples/detectnet.py#L63

I will return the objects detected as a list in the https://github.com/dusty-nv/jetson-inference/blob/627b5890b49449573b7c1af8de22ce985fc395e4/python/examples/detectnet.py#L68

After resolving pyttsx3

I can add in line 70 engine.say(detection) it will speak the name of the objects detected from the class or labels.txt names we have added.

Thanks alot

saifurrehman4114, 2021-06-10:
Hey  @dusty-nv  I have tried using text-to-speech but same issue inside docker please help I have submit my project till 15th june.
**CODE of saif.py:**

from text_to_speech import speak

saif = ['saif']

for i in saif:
 speak(i)

L**### ogs:**

root@fyp-desktop:/jetson-inference/my_project# python3 saif.py 
Traceback (most recent call last):
  File "saif.py", line 6, in <module>
    speak(i)
  File "/usr/local/lib/python3.6/dist-packages/text_to_speech/text_to_speech.py", line 17, in speak
    path = f"C:\\Users\\{os.getlogin()}\\AppData\\Local\\Temp"
OSError: [Errno 6] No such device or address

dusty-nv, 2021-06-10:
Hi @saifurrehman4114, this `text_to_speech` module is unrelated to the jetson-inference project.   You may find better help by posting on the `text_to_speech` github or mailing list or on the Jetson Nano forum.

> File "/usr/local/lib/python3.6/dist-packages/text_to_speech/text_to_speech.py", line 17, in speak
> path = f"C:\Users\{os.getlogin()}\AppData\Local\Temp"

This is a Windows path - it appears this library is for Windows.  Also this pages says Windows OS:  https://pypi.org/project/text-to-speech/

> After resolving pyttsx3
> 
> I can add in line 70 engine.say(detection) it will speak the name of the objects detected from the class or labels.txt names we have added.
> 
> Thanks alot
Have you been able to `import pyttsx3` or not?   Sorry I was confused by your statement above.

saifurrehman4114, 2021-06-10:
I just want the detect objects from SSD to be spoken using any library.

saifurrehman4114, 2021-06-10:
I have tried pyttsx3 too but it's not working 😭

dusty-nv, 2021-06-10:
OK, I was quickly able to install pyttsx3 inside container by running these commands inside container:

```
apt-get update && apt-get install espeak
pip3 install pyttsx3
```

Then I was able to run pyttsx3 without issue.

saifurrehman4114, 2021-06-10:
You can read the thread of pyytsx3 error in but no one responded I am resolving this for the past 3 days unable to resolve & unable to sleep please 🙏 https://forums.developer.nvidia.com/t/how-to-add-voice-capability-during-inference-in-ssd-model/179999/6

dusty-nv, 2021-06-10:
This repo is not about pyttsx3.  I was quickly able to resolve the problem simply by searching the error.  I posted the steps I followed above.  Good luck on your project!

saifurrehman4114, 2021-06-10:
Thank @dusty-nv dusty for your kind wishes and wanted to ask is there any UML model of this repo have tools like Jetson Inferences and Jetson utils which are great came to know about few months.

How like I can intern with Nvidia regarding this matter and also I would like documentation. I have some but things are difficult to find.  Maybe I would like to contribute as an open-source developer.

Any help regarding would be appreciable.

dusty-nv, 2021-06-11:
Hi @saifurrehman4114, you can find the NVIDIA internship listings posted here:   https://www.nvidia.com/en-us/about-nvidia/careers/university-recruiting/

saifurrehman4114, 2021-06-14:
Hey @dusty-nv were you able to run the following code inside the container after installing pyttsx3 as you did in above steps & listen the voice from text to speech conversion. 

Please confirm as I am having error while running this code after following your installation steps of pyttsx3

code :

import pyttsx3 
engine = pyttsx3.init() 
engine.say("I will speak this text") engine.runAndWait()

Thanks @dusty-nv

saifurrehman4114, 2021-06-14:
In detect.py

I have read the docs & I am getting object detected as a str from this line. I just to use some text to speech library to speak that object detected name. 

for detection in detections:
	   print(net.GetClassDesc(detection.Instance))

saifurrehman4114, 2021-06-14:
I have tried changing the library to talkey still the error is there. Unable to find a solution as I have searched in google too. @dusty-nv  any idea how to resolve this error. I am running this code inside container. It's urgent  

Code:
import talkey
tts = talkey.Talkey()
tts.say('Old McDonald had a farm')

Logs
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
aplay: main:788: audio open error: No such file or directory

saifurrehman4114, 2021-06-14:
@dusty-nv I have tried using alsa-utils but still the same error after the installation of pyttsx3 and tall key lib any help to resolve this issue.

dusty-nv, 2021-06-14:
Are you using a USB microphone?

Try adding this line to `docker/run.sh` script here:

`--device /dev/snd --device /dev/bus/usb`
https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/docker/run.sh#L171

Then start container again.

saifurrehman4114, 2021-06-14:
Yes @dusty-nv I am using USB micro phone I added your above mention line still the error is their.

If I do your pyttsx3 steps then this errors shows up

sh: 1: aplay: not found

then to solve this error I got solution of this from Google to install alsa-utils

apt-get install alsa-utils

then I run the basic Pyttsx or tallkey basic program this error shows up

ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
aplay: main:788: audio open error: No such file or directory

These errors are still after I added your line in docker run.sh

dusty-nv, 2021-06-14:
after you run `apt-get install alsa-utils`, if you run `aplay -L` do you see your audio device printed out?

saifurrehman4114, 2021-06-14:
@dusty-nv following error shows up 

 [image]

dusty-nv, 2021-06-14:
After you added `--device /dev/snd --device /dev/bus/usb` to `docker/run.sh` [https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/docker/run.sh#L171](https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/docker/run.sh#L171) (outside of container), did you quit the container and start it again with `docker/run.sh` again?

This is what you should see on Nano (plus the device you plugged in)

```
$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
dmix:CARD=tegrahda,DEV=3
    tegra-hda, HDMI 0
    Direct sample mixing device
dsnoop:CARD=tegrahda,DEV=3
    tegra-hda, HDMI 0
    Direct sample snooping device
hw:CARD=tegrahda,DEV=3
    tegra-hda, HDMI 0
    Direct hardware device without any conversions
plughw:CARD=tegrahda,DEV=3
    tegra-hda, HDMI 0
    Hardware device with all software conversions
default:CARD=tegrasndt210ref
    tegra-snd-t210ref-mobile-rt565x,
    Default Audio Device
sysdefault:CARD=tegrasndt210ref
    tegra-snd-t210ref-mobile-rt565x,
    Default Audio Device
dmix:CARD=tegrasndt210ref,DEV=0
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=1
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=2
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=3
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=4
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=5
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=6
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=7
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=8
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=9
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dsnoop:CARD=tegrasndt210ref,DEV=0
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=1
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=2
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=3
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=4
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=5
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=6
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=7
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=8
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=9
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
hw:CARD=tegrasndt210ref,DEV=0
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
hw:CARD=tegrasndt210ref,DEV=1
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
hw:CARD=tegrasndt210ref,DEV=2
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
hw:CARD=tegrasndt210ref,DEV=3
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
hw:CARD=tegrasndt210ref,DEV=4
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
hw:CARD=tegrasndt210ref,DEV=5
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
hw:CARD=tegrasndt210ref,DEV=6
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
hw:CARD=tegrasndt210ref,DEV=7
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
hw:CARD=tegrasndt210ref,DEV=8
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
hw:CARD=tegrasndt210ref,DEV=9
    tegra-snd-t210ref-mobile-rt565x,
    Direct hardware device without any conversions
plughw:CARD=tegrasndt210ref,DEV=0
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
plughw:CARD=tegrasndt210ref,DEV=1
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
plughw:CARD=tegrasndt210ref,DEV=2
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
plughw:CARD=tegrasndt210ref,DEV=3
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
plughw:CARD=tegrasndt210ref,DEV=4
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
plughw:CARD=tegrasndt210ref,DEV=5
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
plughw:CARD=tegrasndt210ref,DEV=6
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
plughw:CARD=tegrasndt210ref,DEV=7
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
plughw:CARD=tegrasndt210ref,DEV=8
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
plughw:CARD=tegrasndt210ref,DEV=9
    tegra-snd-t210ref-mobile-rt565x,
    Hardware device with all software conversions
```

saifurrehman4114, 2021-06-14:
Yes I did outside the Docker & added that line at line 174.

Still issue is unsolve

Do you Mean to replace 171 line with that or add that after that? that might not be the issue.

As I am mounting a directory to if I add it after line 171 that line I get an that your one argument is misusing.

Is it working for you?

dusty-nv, 2021-06-14:
> Do you Mean to replace 171 line with that or add that after that? that might not be the issue.

Add it after - it should look like this:

```
sudo docker run --runtime nvidia -it --rm --network host -e DISPLAY=$DISPLAY \    
-v /tmp/.X11-unix/:/tmp/.X11-unix \    
-v /tmp/argus_socket:/tmp/argus_socket \    
-v /etc/enctune.conf:/etc/enctune.conf \    
--device /dev/snd --device /dev/bus/usb \
$V4L2_DEVICES $DATA_VOLUME $USER_VOLUME \   
$CONTAINER_IMAGE $USER_COMMAND
```

> Is it working for you?

Yes, after adding `--device /dev/snd --device /dev/bus/usb` above and restarting container, I'm able to see the audio devices inside container with `aplay -L`

saifurrehman4114, 2021-06-14:
Yes Now I am able to see, but when I run the basic pyttsx3 code I see an error which I am unable to remove 

aplay: main:788: audio open error: No such file or directory

Output of
aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
dmix:CARD=tegrahda,DEV=3
    tegra-hda, HDMI 0
    Direct sample mixing device
dsnoop:CARD=tegrahda,DEV=3
    tegra-hda, HDMI 0
    Direct sample snooping device
hw:CARD=tegrahda,DEV=3
    tegra-hda, HDMI 0
    Direct hardware device without any conversions
plughw:CARD=tegrahda,DEV=3
    tegra-hda, HDMI 0
    Hardware device with all software conversions
default:CARD=tegrasndt210ref
    tegra-snd-t210ref-mobile-rt565x,
    Default Audio Device
sysdefault:CARD=tegrasndt210ref
    tegra-snd-t210ref-mobile-rt565x,
    Default Audio Device
dmix:CARD=tegrasndt210ref,DEV=0
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=1
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=2
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=3
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=4
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=5
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=6
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=7
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=8
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dmix:CARD=tegrasndt210ref,DEV=9
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample mixing device
dsnoop:CARD=tegrasndt210ref,DEV=0
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=1
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=2
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=3
    tegra-snd-t210ref-mobile-rt565x,
    Direct sample snooping device
dsnoop:CARD=tegrasndt210ref,DEV=4
    tegra-snd-t210ref-mobile-rt565x,
[... 93 lines cut ...]
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Hardware device with all software conversions

dusty-nv, 2021-06-14:
> Yes @dusty-nv Now I am able to see, but when I run the basic pyttsx3 code I see an error which I am unable to remove
> 
> aplay: main:788: audio open error: No such file or directory

OK, you need to set the default sound device by creating `/etc/asound.conf` inside the container.   Run this inside container:

`printf "defaults.pcm.card 2 \ndefaults.ctl.card 2 \n" > /etc/asound.conf`

Now the way I got the card number `2` was by checking `cat /proc/asound/cards` *outside of container*

```
cat /proc/asound/cards
 0 [tegrahdaxnx    ]: tegra-hda-xnx - tegra-hda-xnx
                      tegra-hda-xnx at 0x3518000 irq 64
 1 [jetsonxaviernxa]: jetson-xaviernx - jetson-xaviernx-ape
                      jetson-xaviernx-ape
 2 [Mono           ]: USB-Audio - Logitech H570e Mono
                      Logitech Inc Logitech H570e Mono at usb-3610000.xhci-2.3, full speed
```

Then I am able to hear the output from pyttsx3

closed 2023-04-27 by dusty-nv