dusty-nv/jetson-inference #1096: TensorRT 8 support

Alex-Beh, 2021-06-10:
Can this repo been used on Ubuntu 20.04 with tensorRT 8?

Thanks for the help.

dusty-nv, 2021-06-10:
Hi, I'm not sure - Jetson doesn't yet support 20.04, and I have only tested this project on Jetson.

Alex-Beh, 2021-06-11:
Hi, thanks for the reply. It seem there are some API changes from tensorRT7 to tensorRT 8.

For example, in tensorRT7([link](https://www.ccoderun.ca/programming/doxygen/tensorrt/classnvinfer1_1_1ILogger.html)):
`virtual void nvinfer1::ILogger::log (Severity severity, const char * msg)
`
But tensorRT8 ([link](https://docs.nvidia.com/deeplearning/tensorrt/api/c_api/classnvinfer1_1_1_i_logger.html)):
`virtual void nvinfer1::ILogger::log (Severity severity, AsciiChar char * msg)
`

Because of the changes, this [function](https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/c/tensorNet.h#L579) will fail during the compilation.

Here is one [example](https://github.com/adujardin/tensorrtx/commit/92eef6353726abba5cbdf4b056377a1e34abd487#diff-63806a07b7b97ced11b7308522a67e70f2cdc6429ba18acc8434776170715564) how to migrate the repo from tensorRT7 to tensorRT8. Hopefully this will help those people who facing the similar issues. 

I hope I can see the Jetson with 20.04 support also. Thanks

closed 2023-04-27 by dusty-nv