dusty-nv/jetson-inference #1097: Use Custom models such as Mobilenetv1

nikunj95, 2021-06-10:
Hi,

Thank you for this repository and youtube videos. I want to train a custom model such as Mobilenet V1 on Jetson nano and I am not sure how to do this here. Could you provide some guidance or direct me to certain links in how to approach this?

Thanks!
Nikunj

dusty-nv, 2021-06-10:
Hi, see this section for ResNet18 and SSD-Mobilenet:   https://github.com/dusty-nv/jetson-inference#training

nikunj95, 2021-06-13:
Thanks for the response. This shows training with PyTorch. I am trying to use Tensorflow. Besides, Mobilenet V1 is not there in Pytorch models. Can you help with working training Tensorflow's model: https://www.tensorflow.org/api_docs/python/tf/keras/applications/MobileNet

closed 2023-04-27 by dusty-nv