dusty-nv/jetson-inference #1098: Retraining ssd-mobilenet support

mabdelbari, 2021-06-10:
Hey,
First of all, thank you for this wonderful tutorial, it helped me a lot. 

I am working on retraining ssd-mobilenet on my own choices of the class list (about 6 classes).

I just wanna understand what --workers is used for and is it fine if I increase it or not?

Thanks in advance

dusty-nv, 2021-06-10:
Hi, workers is the number of dataloader threads. They are only used for loading the images and not the actual training.  I'm not sure if you will see much benefit from increasing it or not.

mabdelbari, 2021-06-10:
Thanks for your response @dusty-nv  😄 

Yea... I got it.

What about increasing batch-size as I wanna increase the speed of training?

dusty-nv, 2021-06-10:
batch size may increase it but if you are on Nano it may be marginal because the GPU is already pretty fully utilized

mabdelbari, 2021-06-11:
Indeed I am working on it from my laptop. So if I increase it, it will result in reducing training time?

dusty-nv, 2021-06-11:
Ah, yes in that case it should

mabdelbari, 2021-06-11:
Thanks for your help. I appreciate it 😃

closed 2021-06-11 by mabdelbari