dusty-nv/jetson-inference #1101: Could not load VGG-16 SSD

anish-faiz, 2021-06-12:
Good afternoon, I am training a custom object detection model using VGG16-SSD, instead of Mobilenet SSD for better accuracy. I have trained the model, but i could not load it. Can you please help me out to load my model and run detection.

dusty-nv, 2021-06-14:
VGG-16 is a classification network, ssd is for object detection.   video streaming is done through imagenet/imagenet.py samples for classification, and detectnet/detectnet.py for object detection.

anish-faiz, 2021-06-14:
Is it possible for me to load VGG16-SSD on detectnet.py
I tried to load but I'm not having skeleton for VGG16-SSD network on detectnet.py
So I have tested with run_ssd_example.py. Now I have to go with video streaming.
Please help me sir...

closed 2023-04-27 by dusty-nv