dusty-nv/jetson-inference #1103: using jetson.utils cuda-tools  downgrade the inference speed

jasonlove521, 2021-06-13:
I do the video detecttion with my own trt-model and use some jetson-utils tools. the code is blow:
//video capture and some preprocesing
 def batchdatagenerate(self, image):
        data = np.empty(
            (0, 3, self.windowsize, self.windowsize), dtype=np.float16)
        image = np.transpose(image, (2, 0, 1)).astype(np.float16)
        image /= 255
        for window in self.windows:
            img_in = image[window.indices()]
            img_in = np.expand_dims(img_in, axis=0)
            img_in = np.ascontiguousarray(img_in)
            data = np.concatenate((data, img_in), axis=0)
        return data.astype(np.float32)

    def update(self):
        while self.started:
            sleep(0.05)
            frame = self.cap.Capture()
            if frame is not None:
                jetson.utils.cudaConvertColor(frame, self.cvt_img)
                jetson.utils.cudaResize(self.cvt_img, self.resized_img)
                data = jetson.utils.cudaToNumpy(self.resized_img)
                frame = jetson.utils.cudaToNumpy(frame)
                data = self.batchdatagenerate(data)
                with self.read_lock:
                    self.frame = frame
                    self.data = data

    def read(self):
        with self.read_lock:
            frame = self.frame.copy()
            data = self.data.copy()
        return frame, data

//inference function
def run(self):
        self.vdo = JetsonCv(self.args)
        self.vdo.start()
        classesname = load_class_names('data/drone.names')
        colors = color(2)
        count = 0
        start = time.time()
        while True:
            im,  data = self.vdo.read()
            count += 1
            self.inputs[0].host = np.ascontiguousarray(data)
            trt_outputs = self.inference_fn(
                context=self.context,
                bindings=self.bindings,
                inputs=self.inputs,
                outputs=self.outputs,
                stream=self.stream)
            trt_outputs[0] = trt_outputs[0].reshape(
                self.batchsize, -1, 1, 4)
            trt_outputs[1] = trt_outputs[1].reshape(
                self.batchsize, -1, self.num_classes)

I just use the jetson.utils.videosource,jetson.utils.cudaConvertColor,jetson.utils.cudaResize,jetson.utils.cudaToNumpy and the use cpu do the others operation,but inference time is increased, i don`t know why. i think that using jeson tools will decrease the inference time, but it is not.

dusty-nv, 2021-06-14:
> I just use the jetson.utils.videosource,jetson.utils.cudaConvertColor,jetson.utils.cudaResize,jetson.utils.cudaToNumpy and the use cpu do the others operation,but inference time is increased, i don`t know why. i think that using jeson tools will decrease the inference time, but it is not.

It's probably because that for such a short simple operation such as cudaConvertColor and cudaResize, the overhead of cudaToNumpy() exceeds this.

dusty-nv, 2021-06-14:
Here is another thing to try - cudaToNumpy() mappings are persistent, so you only need to do them once per buffer.  So you could call it once for your `self.resized_img` buffer and then re-use it the rest of your program.

jasonlove521, 2021-06-15:
@dusty-nv Thanks for your reply,I will try it.

closed 2023-04-27 by dusty-nv