dusty-nv/jetson-inference #1104: Easy way to install pyTorch with  Gpu /Cuda support on Jetson nano

SaddamBInSyed, 2021-06-14:
@dusty-nv 

I have my new Jetson nano device here and would like to install PyTorch with GPU support.

Can you please advise the easiest way to install PyTorch with  GPU / Cuda support on Jetson nano?

I have tried several methods but no luck still ?

thank you

dusty-nv, 2021-06-14:
If you run the jetson-inference container, it already has PyTorch with GPU installed in the container.

For the PyTorch wheels, see this thread:  https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-8-0-now-available/72048

SaddamBInSyed, 2021-06-14:
@dusty-nv 
S. I have followed this "https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-8-0-now-available/72048" thread .

now torch running on my cuda.

Thank you

closed 2021-06-14 by SaddamBInSyed