dusty-nv/jetson-inference #1109: custom caffe model convert to tensorrt issues

SokPhanith, 2021-06-16:
Hello @dusty-nv  , I train on my own custom data-set Image Classification 4 classes with caffe. I copy from  path : jetson-inference/build/aarch64/bin/networks/ResNet-18/ file below:
1- first train : 
- train.prototxt : delete acc/top-5 layer , make some edit path data-set lmdb format and  
    transform_param {
        mirror: true
        crop_size: 224
        mean_value: 104
        mean_value: 117
        mean_value: 123
    }  
- solver.prototxt : edit some path
- deploy.prototxt : edit output classes

I build tensorrt and run inference working good. but 
2- second train : 
- train.prototxt : delete acc/top-5 layer , make some edit path data-set lmdb format and  
  transform_param {
    mirror: true
    crop_size: 224
    mean_file: "/home/phanith/Desktop/resnet18/board/mean.binaryproto"
    }
I compute mean file from caffe tool on my custom data-set
- solver.prototxt : edit some path
- deploy.prototxt : edit output classes
I build tensorrt and run inference working not good. very low Classification

I want to ask you when tensorrt Image Classification inference with caffe model format transform_param default look like ?
if I have a mean.binaryproto file compute by caffe tool how can I set path for inference. I really not sure for it when tensorrt inference need or not. By the way 2 training I test on my validation set got accuracy around 98% the same.

dusty-nv, 2021-06-16:
Hi @SokPhanith, jetson-inference doesn't support loading the mean.binaryproto file - if needed, you can change the average mean pixel value here:   https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/c/imageNet.cpp#L449

However this does not seem necessary if you are still getting 98% accuracy

SokPhanith, 2021-06-16:
Thank you @dusty-nv. Sorry, Can you give me some example add move code to load mean.binaryproto file ?

dusty-nv, 2021-06-16:
Sorry, I don't have the code for doing that or using the per-pixel mean data in the preprocessing