dusty-nv/jetson-inference #1111: Can't download pre-train Inception-v4 caffe

SokPhanith, 2021-06-17:
sorry when I run ./download-models.sh and then space to Inception-v4 for download but the script skip for download this model.
from : https://github.com/SnailTyan/caffe-model-zoo/tree/master/Inception-v4 I can't find pre-trian too.

dusty-nv, 2021-06-17:
Hi @SokPhanith, you can find it on this page also:  https://github.com/dusty-nv/jetson-inference/releases/tag/model-mirror-190618

SokPhanith, 2021-06-17:
Thank you so much @dusty-nv

SokPhanith, 2021-06-17:
sorry, How can I download train_val.prototxt and solver.prototxt with model : resnet50 resnet101 resnet152 that using train for caffe. I can't fine at path : jetson-inference/data/networks

dusty-nv, 2021-06-17:
I believe these models originally came from here:  https://github.com/KaimingHe/deep-residual-networks

There appears to only be the deploy prototxt

closed 2023-04-27 by dusty-nv