dusty-nv/jetson-inference #1112: cannot run eval_ssd.py,  due to load_state_dict?

WenY2020, 2021-06-17:
Hello Dusty:

Thanks a lot for sharing your work and I am trying to learn and apply it.

But I have an issue when i ran eval_ssd.py, 
actually I have trained the detection model on my own dataset, and got the checkpoints  e.g. mb1-ssd-Epoch-94-Loss-2.581278615527683.pth, but when i ran eval_ssd.py on this checkpoint, the error of 'load_state_dict' comes up like below:

Can you please give a hint or idea how to check this up? thanks so much!

[image]

dusty-nv, 2021-06-17:
Hi @WenY2020, can you test that you can run `run_ssd_example.py` and `export_onnx.py` on the .pth checkpoint?  If not, maybe the checkpoint is corrupt

WenY2020, 2021-06-23:
Hello, Dusty, thanks a lot for your reply!

Actually export_onnx.py always works for me, there is no problem with that.

And I have found the reason why I got the error with running eval_ssd.py: the default net in eval_ssd.py is vgg-16, but the model trained (and also pretrained) is mb1-ssd, so we need to set the net be mb1-ssd, this is the main reason that the error comes from,  and also share with you or other people who may be interested: (the below works for me)
[image]

Thanks a lot for your time!

Best wishes,
Wen:)

closed 2021-07-08 by WenY2020