dusty-nv/jetson-inference #1113: Train on Mobilenet V2 instead of V1 and aspect ratio?

stekla79, 2021-06-17:
Hi, 
I have a few questions regarding retraining on the SSD-mobilenet. 

1. In the examples the pretrained weights is downloaded from the command below. 
```wget https://nvidia.box.com/shared/static/djf5w54rjvpqocsiztzaandq1m3avr7c.pth -O models/mobilenet-v1-ssd-mp-0_675.pth```
This is downloadingen the pretrained weights for mobilenet v1? Is there also a link to for downloading mobilenet v2?

2. With the training data I have images in different sizes and also some differences in aspect ratio in the images. Some is 16:9, some is 4:3, some is horizontal and some is potrait etc. Should I take any special care in rezising the images before training? The training script automatically resizes the images to 300x300 automatically? Would any preproccessing help or aid?

dusty-nv, 2021-06-17:
Hi @stekla79, I haven't tested mobilenet-v2 with this repo so am unsure the steps needed.   The main enhancement in mobilenet-v2 is the use of separable convolutions for mobile platforms (i.e. phones), but this is not needed for Jetson and ssd-mobilenet-v1 is fine.

The images are automatically resized to 300x300 by PyTorch during the training, so you don't need to worry about it.  There is also data augmentation that automatically occurs.

fiskdata, 2021-06-17:
Alright, thanks for answering my questions!

stekla79, 2021-06-17:
Alright, Thanks for answering the questions.

closed 2021-06-17 by stekla79