dusty-nv/jetson-inference #1114: I am getting "ERROR: Unexpected segmentation fault encountered in worker."  in the middle of the training of a detection model.

Eduardo140297, 2021-06-17:
The training stopped in the middle of epoch 25 (out of 30) and I got this error message:

ERROR: Unexpected segmentation fault encountered in worker.
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 779, in _try_get_data
    data = self._data_queue.get(timeout=timeout)
  File "/usr/lib/python3.6/multiprocessing/queues.py", line 104, in get
    if not self._poll(timeout):
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 257, in poll
    return self._poll(timeout)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 414, in _poll
    r = wait([self], timeout)
  File "/usr/lib/python3.6/multiprocessing/connection.py", line 911, in wait
    ready = selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 376, in select
    fd_event_list = self._poll.poll(timeout)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/_utils/signal_handling.py", line 66, in handler
    _error_if_any_worker_fails()
RuntimeError: DataLoader worker (pid 380) is killed by signal: Segmentation fault.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "train_ssd.py", line 343, in <module>
    device=DEVICE, debug_steps=args.debug_steps, epoch=epoch)
  File "train_ssd.py", line 113, in train
    for i, data in enumerate(loader):
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 363, in __next__
    data = self._next_data()
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 974, in _next_data
    idx, data = self._get_data()
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 941, in _get_data
    success, data = self._try_get_data()
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 792, in _try_get_data
    raise RuntimeError('DataLoader worker (pid(s) {}) exited unexpectedly'.format(pids_str))
RuntimeError: DataLoader worker (pid(s) 380) exited unexpectedly

why is this happening?

fivefishstudios, 2021-09-22:
I'm getting an immediate error "Segmentation fault (core dumped)" trying to run the train_ssd.py... No traceback messages, etc. 

I have a 2GB Nano Dev Kit. Is this train_ssd.py program for the 4GB Nano only?

dusty-nv, 2021-09-22:
It should work on Nano 2GB as well as Nano 4GB.

Are you running the docker container?

What if you run "export OPENBLAS_CORETYPE=ARMV8" before hand?

https://github.com/numpy/numpy/issues/18131#issuecomment-756140369

fivefishstudios, 2021-09-23:
Hi Dusty

Doing the export OPENBLAS didn't do anything, same error message. 

I looked at the source code of train_ssd.py and found the line where Python3 is complaining. 

Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> import sys
>>> import logging
>>> import argparse
>>> import itertools
>>> import torch
>>> from torch.utils.data import DataLoader, ConcatDataset
>>> from torch.optim.lr_scheduler import CosineAnnealingLR, MultiStepLR
>>> from vision.utils.misc import str2bool, Timer, freeze_net_layers, store_labels
>>> from vision.ssd.ssd import MatchPrior
>>> from vision.ssd.vgg_ssd import create_vgg_ssd
**Segmentation fault (core dumped)**

Any thoughts on why "import create_vgg_ssd" is causing the segmentation fault? 

Thank you
Ruel

dusty-nv, 2021-09-23:
Hi Ruel, are you using the jetson-inference container?  If not, are you able to try the container and see if it has the same issue?  That could narrow down if it is related to your PyTorch install or not (although can't recall seeing this error before)

fivefishstudios, 2021-09-23:
Hi Dusty, I'm running on the Nano itself (not a container).
BUT... I manage to solve the problem and got train_ssd.py running.  I had to do this:

**export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1** 

Found this tidbit on the nVidia forum. I encountered a new error message
Cv2: Cannot allocate memory in static TLS block
and googling that led to the nVidia forum with the export command as the fix. 

Will report back if it finished training completely. 

Thanks again! 
Ruel