dusty-nv/jetson-inference #1115: Autocomplete not working for VScode or Pycharm

sardanian, 2021-06-18:
I am trying to use autocomplete in VSCode or Pycharm and am unable to. I am using the docker container and am connecting to the docker container via VSCode. I can run my code within VSCode fine, however I do not get any benefit of autocomplete.

I have seen this issue before with CV2 not having autocomplete but a step like this below link shows solves that:
https://spyjetson.blogspot.com/2020/

Is there any way to have intellisense for development I am also ok using different IDEs, although I would prefer VSCode or Pycharm.

dusty-nv, 2021-06-18:
Hi @sardanian I haven't used vscode but my guess is that intellisense/autocomplete is not working because the jetson-inference Python libraries are C extension modules and not actual Python code.

sezer-muhammed, 2022-01-01:
I also need that too, as far as I know, there is a way to make it work but I don't know how. @dusty-nv what is your ide and can it complete automaticly?

dusty-nv, 2022-01-04:
@sezer-muhammed I just use a text editor and command line

closed 2023-04-27 by dusty-nv