dusty-nv/jetson-inference #1117: How do I use .onnx to webcam object detection?

ttaehyung, 2021-06-18:
Hi, I'm currently working on my object-detection project with Jetson Nano.
I trained my yolov4-tiny model using darknet, with my own dataset.
With darknet2ONNX (https://github.com/Tianxiaomo/pytorch-YOLOv4#3-darknet2onnx) I was able to create ONNX file from my trained weightfile and cfg file.
After that, I tried to launch real-time object detection display with detectnet.py or detectnet-camera.py or my own detectnet code with jetson.inference module.

Execution code with original file was:

detectnet.py --model=yolov4_1_3_416_416_static.onnx --labels=obj.txt --input-blob=input --output-cvg=confs --output-bbox=boxes /dev/video1

My own detectnet code was:
[image]

However, my network cannot detect any objects!!
[image]

I'm really stuck here so please help T.T

dusty-nv, 2021-06-18:
Hi @ttaehyung, this repo is setup for SSD-Mobilenet ONNX for detection models that use ONNX.  You would need to make sure that the pre/post-processing match what the YOLO ONNX model expects.

* Pre-processing: https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/c/detectNet.cpp#L729
* Post-processing: https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/c/detectNet.cpp#L813

If you are unable to get it working, you may just want to add cv2.VideoCapture() code to the pytorch-YOLOv4 code so that it can process webcam

ttaehyung, 2021-06-18:
Thanks, but I'm pretty new to this stuff(code, jetson nano...) I don't know how should I change them.
My model is yolov4-tiny, so how do I import the model?

dusty-nv, 2021-06-18:
I haven't used YOLOv4-tiny with this project before - new models typically require the custom pre/post-processing

You might want to check this blog for running YOLOv4 through TensorRT:  https://jkjung-avt.github.io/tensorrt-yolov4/

ttaehyung, 2021-06-18:
Thanks a lot! I'll take a look.

closed 2021-06-18 by ttaehyung