dusty-nv/jetson-inference #1118: What is the best way to integrate jetson inference with opencv

gitgkk, 2021-06-18:
Hi @dusty-nv,

I'm unable to find an answer to my question. I know that OpenCV can be compiled and installed with Cuda support on jetson, I've done that before. Is this the recommended way of using opencv on jetson?

OR

Do you suggest that openCV with Cuda support be installed in the jetson inference docker? If yes, what is the procedure?
What else do I need?

Thanks for your support.

Thanks,
Kashyap

dusty-nv, 2021-06-18:
If you don't want to use container, you can just build/install OpenCV w/ CUDA outside of container.

If you want to use container, then you can add the steps to the Dockerfile.   Or perhaps you are able to build deb packages for OpenCV outside of container, and then just install those deb packages into the container.

As far as integration goes, jetson-inference has cudaToNumpy() and cudaFromNumpy() for interoperability with OpenCV:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-to-numpy-arrays
* https://github.com/dusty-nv/jetson-utils/blob/ebab1914877a51d4d33fa9b1f01b168adb712a32/python/examples/cuda-to-cv.py

gitgkk, 2021-06-19:
Hi @dusty-nv 

Thanks for the response. I thought the same.

For the scenario when OpenCV is built with CUDA outside the container. Will I be able to use jetson inference library and how?
Can you point me to some examples?

Where can I find the docker file and information on how to modify it for installing compiled OpenCV with CUDA installation?

Thanks

dusty-nv, 2021-06-21:
@gitgkk here is a script that can build OpenCV with CUDA:

https://github.com/mdegans/nano_build_opencv

You would essentially add this to the jetson-inference dockerfile:  https://github.com/dusty-nv/jetson-inference/blob/master/Dockerfile

gitgkk, 2021-06-21:
Hi @dusty-nv, you're a great help! Wonder what would the community do without you. Thanks for sharing the links.

closed 2023-04-27 by dusty-nv