dusty-nv/jetson-inference #1120: compile jetson inference error cmake

syliabe, 2021-06-21:
Hello, 
I am trying to run jetson inference cmake ../ I get this error

thanks
[image]

dusty-nv, 2021-06-21:
Hi @syliabe, it doesn't look like your Jetson was setup with CUDA Toolkit (and probably not TensorRT either)

If you are using the Nano or Xavier NX devkit, CUDA should already come pre-installed on the SD card images.

If you are using another Jetson that you flashed with SDK Manager, run SDK Manager again from your PC and have it install CUDA, cuDNN, TensorRT, ect to your Jetson device.

syliabe, 2021-06-21:
sorry I did not specify that I am working on an x64  pc

syliabe, 2021-06-21:
is there a specific version (CUDA Toolkit / cuDNN / TensoRrt? or any?

dusty-nv, 2021-06-21:
I would suggest CUDA 10.2, cuDNN 8, and TensorRT 7.1 because this is what the Jetson currently runs in JetPack 4.5 (and hence those versions I have tested)

closed 2023-04-27 by dusty-nv