dusty-nv/jetson-inference #1121: Can't build ssd-mobilenet caffe model to tensorrt

SokPhanith, 2021-06-22:
Hello @dusty-nv .
I train my custom dataset object detection caffe model SSD-Moblienet from https://github.com/chuanqi305/MobileNet-SSD.
I test a simple is working for me. but when I try build tensorrt with command line didn't work.

 /detectnet-camera --model=/home/phanith/MobileNet-SSD/dataset/output/mobilenet_iter_3500.caffemodel --prototxt=/home/phanith/MobileNet-SSD/dataset/prototxt/MobileNetSSD_deploy.prototxt --labels=/home/phanith/MobileNet-SSD/dataset/list/labels.txt --input-blob="data" --output-bbox="detection_out" --mean-pixel=127.5 csi://0 --input-flip=rotate-180

-Error:

Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
Warning: Flatten layer ignored. TensorRT implicitly flattens input to FullyConnected layers, but in other circumstances this will result in undefined behavior.
[TRT]    mbox_loc: all concat input tensors must have the same dimensions except on the concatenation axis (0), but dimensions mismatched at index 1. Input 0 shape: [19,19,12], Input 1 shape: [10,10,24]
[TRT]    mbox_loc: all concat input tensors must have the same dimensions except on the concatenation axis (0), but dimensions mismatched at index 1. Input 0 shape: [19,19,12], Input 1 shape: [10,10,24]
Caffe Parser: Invalid axis in softmax layer - TensorRT expects NCHW input. Negative axis is not supported in TensorRT, please use positive axis indexing
error parsing layer type Softmax index 186
[TRT]    device GPU, failed to parse caffe network
[TRT]    device GPU, failed to load /home/phanith/MobileNet-SSD/dataset/output/mobilenet_iter_3500.caffemodel
[TRT]    detectNet -- failed to initialize.
detectnet:  failed to load detectNet model

dusty-nv, 2021-06-22:
Hi @SokPhanith, I haven't tried SSD-Mobilenet through Caffe before.  I recommend to use this PyTorch version here, which is known to be working:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md

SokPhanith, 2021-06-23:
I used to Re-train on it https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md but I saw /detectnet-camera support with caffe model too. Can you give me Github source or caffe model that SSD-Mobilenet can build with  /detectnet-camera

dusty-nv, 2021-06-23:
That is only the older 'detectNet' DNN architecture trained with DIGITS that worked with caffe.   The newer SSD-Mobilenet model that works with TensorRT is trained with PyTorch.  I don't know of SSD-Mobilenet from Caffe that works with TensorRT.

closed 2023-04-27 by dusty-nv