dusty-nv/jetson-inference #1122: Open Images Dataset

yigittopcu, 2021-06-22:
Hello,

Open images dataset doesn't have the category I have been looking for what kind of dataset can I use in the tutorial ? For training and converting to onnx later.

Thanks.

dusty-nv, 2021-06-22:
Hi @yigittopcu, you can collect your own data following this tutorial:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md

Or if you can find images of the objects, you can annotate them with CVAT tool (CVAT.org) and then export them in Pascal VOC format.

yigittopcu, 2021-06-22:
Thanks for quick reply @dusty-nv . I have one more question. I already have Pascal VOC XML dataset. Can I use it directly ? Or Do I need to convert it ?

dusty-nv, 2021-06-22:
@yigittopcu you can use it as long as the directory structure is the same as the original Pascal VOC dataset.  If not, make the directory structure the same as Pascal VOC

yigittopcu, 2021-06-29:
@dusty-nv  I did what you told me and trained with Pascal Voc dataset. I don't understand what is wrong but there is a example of the problem below . 

[image]

dusty-nv, 2021-06-29:
Hi @yigittopcu, is this using jetson.inference.detectNet to run the model?  If so, you may lower the clustering threshold here:

https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/c/detectNet.h#L477

After changing this, re-run `make` and `sudo make install`

yigittopcu, 2021-06-29:
> Hi @dusty-nv . This is deepstream. I changed pre-clusterthreshold to 1 from 0.2 and boxes all gone. Any value below 1 results same output. 

Is there possibility this caused by training ?

dusty-nv, 2021-06-29:
> Is there possibility this caused by training ?

You can test your model on the PyTorch side first using the [`run_ssd_example.py`](https://github.com/dusty-nv/pytorch-ssd/blob/8ed842a408f8c4a8812f430cf8063e0b93a56803/run_ssd_example.py) script.  Use one of your PyTorch checkpoints with this script, not the ONNX model.

yigittopcu, 2021-06-29:
`python3 run_ssd_example.py mb1-ssd models/Hardhat/mb1-ssd-Epoch-99-Loss-1.7641232138971639.pth models/Hardhat/labels.txt ~/home/netcoz/pytorch-ssd/data/Hardhat/JPEGImages/000010_jpg.rf.ff3d356d21409f8e9fe185536c4ecad9.jpg
`
Got this error with run_ssd_example.py

Traceback (most recent call last):
  File "run_ssd_example.py", line 50, in <module>
    image = cv2.cvtColor(orig_image, cv2.COLOR_BGR2RGB)
cv2.error: OpenCV(4.5.2) /tmp/pip-req-build-wu1ri_rx/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

dusty-nv, 2021-06-29:
I believe it failed to load that image

yigittopcu, 2021-06-29:
Thanks. I will try again first thing tomorrow.

yigittopcu, 2021-06-30:
```
Inference time:  0.005075693130493164
Traceback (most recent call last):
  File "run_ssd_example.py", line 55, in <module>
    cv2.rectangle(orig_image, (box[0], box[1]), (box[2], box[3]), (255, 255, 0), 4)
cv2.error: OpenCV(4.5.2) :-1: error: (-5:Bad argument) in function 'rectangle'
> Overload resolution failed:
>  - Can't parse 'pt1'. Sequence item with index 0 has a wrong type
>  - Can't parse 'pt1'. Sequence item with index 0 has a wrong type
>  - Can't parse 'rec'. Expected sequence length 4, got 2
>  - Can't parse 'rec'. Expected sequence length 4, got 2

```

closed 2023-04-27 by dusty-nv