dusty-nv/jetson-inference #1124: compilation problems MAKE

syliabe, 2021-06-23:
Hello, 
please I am trying to compile the project but I am having problems; When I do make I get this error:

`-- Copying examples/segnet.py -> segnet-camera.py
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sylia/jetson-inference/build
[ 48%] Built target jetson-utils
[ 49%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o
/home/sylia/jetson-inference/c/tensorNet.h(610): error: exception specification for virtual function "tensorNet::Profiler::reportLayerTime" is incompatible with that of overridden function "nvinfer1::IProfiler::reportLayerTime"

/home/sylia/jetson-inference/c/tensorNet.h(579): error: exception specification for virtual function "tensorNet::Logger::log" is incompatible with that of overridden function "nvinfer1::ILogger::log"

2 errors detected in the compilation of "/home/sylia/jetson-inference/c/detectNet.cu".
CMake Error at jetson-inference_generated_detectNet.cu.o.cmake:279 (message):
  Error generating file
  /home/sylia/jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_detectNet.cu.o

CMakeFiles/jetson-inference.dir/build.make:335: recipe for target 'CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
`

thanks

dusty-nv, 2021-06-23:
Hi @syliabe, it looks like you are compiling on x86, and I have only built this repo on Jetson.

My guess is you have a different version of TensorRT installed than I have tested against.  I have tested up to TensorRT 7.1 because that is the latest that JetPack is on for Jetson.

syliabe, 2021-06-23:
Yes I am working on the PC,

But the problem a few months ago it worked on my I was able to do the training and test them!
and when I reset everything, I get this error!

dusty-nv, 2021-06-23:
Which version of TensorRT do you have installed now?

syliabe, 2021-06-23:
tensorRT8.0.0

dusty-nv, 2021-06-23:
I haven't tested against TensorRT 8.0 yet, because the latest Jetson has is TensorRT 7.1.  So you may want to downgrade to TensorRT 7.1.

syliabe, 2021-06-23:
i will test it, thanks

syliabe, 2021-06-24:
to give you a feedback, the project works using version cuda10.2 and TensorRt 7
thank you

closed 2021-06-24 by syliabe

surefyyq, 2021-12-27:
Hi @dusty-nv , at this time the latest JetPack4.6(rev.3) includes TensorRT 8.0.1 and the same error appears when making the  repo on tx2.

> /home/tx2/jetson-inference/c/tensorNet.h(610): error: exception specification for virtual function "tensorNet::Profiler::reportLayerTime" is incompatible with that of overridden function "nvinfer1::IProfiler::reportLayerTime"
/home/tx2/jetson-inference/c/tensorNet.h(579): error: exception specification for virtual function "tensorNet::Logger::log" is incompatible with that of overridden function "nvinfer1::ILogger::log"
2 errors detected in the compilation of "/tmp/tmpxft_00004c70_00000000-8_segNet.compute_72.cpp1.ii".
CMake Error at jetson-inference_generated_segNet.cu.o.cmake:279 (message):
  Error generating file
  /home/tx2/jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_segNet.cu.o
CMakeFiles/jetson-inference.dir/build.make:70: recipe for target 'CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_segNet.cu.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_segNet.cu.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Should I have to roll back to JetPack 4.5 or 4.4? 
Can I make some modification in the TensorRT 8.0.1 environment without re-flashing the tx2?
Thank you!

AndreeaVM, 2022-09-07:
Hi!
I also have this problem when I try to run the docker image for Jetpack 5.0 and TensorRT8.4 on x86 host.
Please, what should I do?

[ 75%] Linking CXX shared library ../../../x86_64/lib/python/3.8/jetson_utils_python.so
[ 75%] Built target video-viewer
[ 75%] Linking CXX executable ../../../x86_64/bin/camera-viewer
[ 75%] Built target jetson-utils-python-38
[ 75%] Built target camera-viewer
/jetson-inference/c/tensorNet.h(617): error: exception specification for virtual function "tensorNet::Profiler::reportLayerTime" is incompatible with that of overridden function "nvinfer1::IProfiler::reportLayerTime"

/jetson-inference/c/tensorNet.h(586): error: exception specification for virtual function "tensorNet::Logger::log" is incompatible with that of overridden function "nvinfer1::ILogger::log"

2 errors detected in the compilation of "/jetson-inference/c/segNet.cu".
CMake Error at jetson-inference_generated_segNet.cu.o.cmake:279 (message):
  Error generating file
  /jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_segNet.cu.o

make[2]: *** [CMakeFiles/jetson-inference.dir/build.make:84: CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_segNet.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/jetson-inference/c/tensorNet.h(617): error: exception specification for virtual function "tensorNet::Profiler::reportLayerTime" is incompatible with that of overridden function "nvinfer1::IProfiler::reportLayerTime"

/jetson-inference/c/tensorNet.h(586): error: exception specification for virtual function "tensorNet::Logger::log" is incompatible with that of overridden function "nvinfer1::ILogger::log"

[ 75%] Building CXX object utils/display/gl-display-test/CMakeFiles/gl-display-test.dir/gl-display-test.cpp.o
In file included from /jetson-inference/build/x86_64/include/jetson-utils/imageFormat.h:240,
                 from /jetson-inference/build/x86_64/include/jetson-utils/videoOptions.h:26,
                 from /jetson-inference/build/x86_64/include/jetson-utils/videoOutput.h:27,
                 from /jetson-inference/build/x86_64/include/jetson-utils/glDisplay.h:26,
                 from /jetson-inference/utils/display/gl-display-test/gl-display-test.cpp:23:
/jetson-inference/build/x86_64/include/jetson-utils/imageFormat.inl: In function ‘imageFormat imageFormatFromType()’:
/jetson-inference/build/x86_64/include/jetson-utils/imageFormat.inl:235:1: warning: no return statement in function returning non-void [-Wreturn-type]
  235 | }
      | ^
[ 76%] Linking CXX executable ../../../x86_64/bin/gl-display-test
2 errors detected in the compilation of "/jetson-inference/c/detectNet.cu".
CMake Error at jetson-inference_generated_detectNet.cu.o.cmake:279 (message):
  Error generating file
  /jetson-inference/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_detectNet.cu.o

make[2]: *** [CMakeFiles/jetson-inference.dir/build.make:77: CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o] Error 1
[ 76%] Built target gl-display-test
/jetson-inference/build/x86_64/include/jetson-utils/imageFormat.inl: In function ‘imageFormat imageFormatFromType()’:
/jetson-inference/build/x86_64/include/jetson-utils/imageFormat.inl:235:1: warning: no return statement in function returning non-void [-Wreturn-type]
  235 | }
      | ^
make[1]: *** [CMakeFiles/Makefile2:436: CMakeFiles/jetson-inference.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
The command '/bin/sh -c mkdir docs &&     touch docs/CMakeLists.txt &&     sed -i 's/nvcaffe_parser/nvparsers/g' CMakeLists.txt &&     mkdir build &&     cd build &&     cmake ../ &&     pwd &&     ls -al &&     make -j${NPROC_BUILD} &&     make install &&     rm -rf /var/lib/apt/lists/*' returned a non-zero code: 2

Thank you!