dusty-nv/jetson-inference #1125: Inference detectnet

syliabe, 2021-06-24:
Hello, 
Please, I am trying to inference with detecnet and my network retrained, I am receiving these errors. 

`[image] loaded 'data/images/humans_0.jpg'  (413x300, 3 channels)
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/tensorConvert.cu:236
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/detectNet.cpp:734
[TRT]    detectNet::Detect() -- cudaTensorNormMeanRGB() failed
[OpenGL] glDisplay -- set the window size to 413x300
[OpenGL] creating 413x300 texture (GL_RGB8 format, 371700 bytes)
[cuda]   registered openGL texture for interop access (413x300, GL_RGB8, 371700 bytes)
[image] saved 'data/test/humans_0.jpg'  (413x300, 3 channels)

[TRT]    ------------------------------------------------
[TRT]    Timing Report models/Person/ssd-mobilenet.onnx
[TRT]    ------------------------------------------------
[cuda]      invalid resource handle (error 400) (hex 0x190)
[cuda]      /home/sylia/Desktop/jetson-inference/build/x86_64/include/jetson-inference/tensorNet.h:685
[TRT]    Pre-Process   CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   0.00000ms  CUDA   0.00000ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[image] loaded 'data/images/humans_1.jpg'  (585x388, 3 channels)
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/tensorConvert.cu:236
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/detectNet.cpp:734
[TRT]    detectNet::Detect() -- cudaTensorNormMeanRGB() failed
[OpenGL] glDisplay -- set the window size to 585x388
[OpenGL] creating 585x388 texture (GL_RGB8 format, 680940 bytes)
[cuda]   registered openGL texture for interop access (585x388, GL_RGB8, 680940 bytes)
[image] saved 'data/test/humans_1.jpg'  (585x388, 3 channels)

[TRT]    ------------------------------------------------
[TRT]    Timing Report models/Person/ssd-mobilenet.onnx
[TRT]    ------------------------------------------------
[cuda]      invalid resource handle (error 400) (hex 0x190)
[cuda]      /home/sylia/Desktop/jetson-inference/build/x86_64/include/jetson-inference/tensorNet.h:685
[TRT]    Pre-Process   CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   0.00000ms  CUDA   0.00000ms
[TRT]    ------------------------------------------------

[image] loaded 'data/images/humans_2.jpg'  (1024x669, 3 channels)
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/tensorConvert.cu:236
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/detectNet.cpp:734
[TRT]    detectNet::Detect() -- cudaTensorNormMeanRGB() failed
[OpenGL] glDisplay -- set the window size to 1024x669
[OpenGL] creating 1024x669 texture (GL_RGB8 format, 2055168 bytes)
[cuda]   registered openGL texture for interop access (1024x669, GL_RGB8, 2055168 bytes)
[image] saved 'data/test/humans_2.jpg'  (1024x669, 3 channels)

[TRT]    ------------------------------------------------
[TRT]    Timing Report models/Person/ssd-mobilenet.onnx
[TRT]    ------------------------------------------------
[cuda]      invalid resource handle (error 400) (hex 0x190)
[cuda]      /home/sylia/Desktop/jetson-inference/build/x86_64/include/jetson-inference/tensorNet.h:685
[TRT]    Pre-Process   CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   0.00000ms  CUDA   0.00000ms
[TRT]    ------------------------------------------------

[image] loaded 'data/images/humans_3.jpg'  (1000x667, 3 channels)
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/tensorConvert.cu:236
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/detectNet.cpp:734
[TRT]    detectNet::Detect() -- cudaTensorNormMeanRGB() failed
[OpenGL] glDisplay -- set the window size to 1000x667
[OpenGL] creating 1000x667 texture (GL_RGB8 format, 2001000 bytes)
[cuda]   registered openGL texture for interop access (1000x667, GL_RGB8, 2001000 bytes)
[image] saved 'data/test/humans_3.jpg'  (1000x667, 3 channels)

[TRT]    ------------------------------------------------
[TRT]    Timing Report models/Person/ssd-mobilenet.onnx
[TRT]    ------------------------------------------------
[cuda]      invalid resource handle (error 400) (hex 0x190)
[cuda]      /home/sylia/Desktop/jetson-inference/build/x86_64/include/jetson-inference/tensorNet.h:685
[TRT]    Pre-Process   CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   0.00000ms  CUDA   0.00000ms
[TRT]    ------------------------------------------------

[image] loaded 'data/images/humans_4.jpg'  (1000x667, 3 channels)
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/tensorConvert.cu:236
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/detectNet.cpp:734
[TRT]    detectNet::Detect() -- cudaTensorNormMeanRGB() failed
[image] saved 'data/test/humans_4.jpg'  (1000x667, 3 channels)

[TRT]    ------------------------------------------------
[TRT]    Timing Report models/Person/ssd-mobilenet.onnx
[TRT]    ------------------------------------------------
[cuda]      invalid resource handle (error 400) (hex 0x190)
[cuda]      /home/sylia/Desktop/jetson-inference/build/x86_64/include/jetson-inference/tensorNet.h:685
[TRT]    Pre-Process   CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   0.00000ms  CUDA   0.00000ms
[TRT]    ------------------------------------------------

[image] loaded 'data/images/humans_5.jpg'  (1024x606, 3 channels)
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/tensorConvert.cu:236
[cuda]      no kernel image is available for execution on the device (error 209) (hex 0xD1)
[cuda]      /home/sylia/Desktop/jetson-inference/c/detectNet.cpp:734
[TRT]    detectNet::Detect() -- cudaTensorNormMeanRGB() failed
[OpenGL] creating 1024x606 texture (GL_RGB8 format, 1861632 bytes)
[cuda]   registered openGL texture for interop access (1024x606, GL_RGB8, 1861632 bytes)
[image] saved 'data/test/humans_5.jpg'  (1024x606, 3 channels)

[TRT]    ------------------------------------------------
[TRT]    Timing Report models/Person/ssd-mobilenet.onnx
[TRT]    ------------------------------------------------
[cuda]      invalid resource handle (error 400) (hex 0x190)
[cuda]      /home/sylia/Desktop/jetson-inference/build/x86_64/include/jetson-inference/tensorNet.h:685
[TRT]    Pre-Process   CPU   0.00000ms  CUDA   0.00000ms
[TRT]    Total         CPU   0.00000ms  CUDA   0.00000ms
[TRT]    ------------------------------------------------

[image] imageLoader -- End of Stream (EOS) has been reached, stream has been closed
detectnet:  shutting down...
detectnet:  shutdown complete.`

Suggestions please thank you

dusty-nv, 2021-06-24:
Hi @syliabe, if you are running on a non-Jetson device, you need to add your GPU's compute capability to CMakeLists.txt here, and rebuild/reinstall:

https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/CMakeLists.txt#L63

syliabe, 2021-06-25:
Thank you, that was it, well corrected

closed 2023-04-27 by dusty-nv