dusty-nv/jetson-inference #1129: Bird detection on a live camera

Fatemehshr, 2021-06-26:
I am following Dusty’s tutorials about using Jetson-inference and I have run 10 lines of your Python code. but I want to detect only birds in real time. How do I change Detectnet.py for detecting only birds?

dusty-nv, 2021-06-28:
Hi @Fatemehshr, if you change the class names in the model's `labels.txt` that you want to ignore to `void`, it will ignore them during detection.

saifurrehman4114, 2021-06-28:
https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet.py#L69

you can also try this . I am ignoring the background after detection 

`if (net.GetClassDesc(detection.Instance) != 'BACKGROUND'):`

Fatemehshr, 2021-06-28:
> 
> 
> https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet.py#L69
> 
> you can also try this . I am ignoring the background after detection
> 
> `if (net.GetClassDesc(detection.Instance) != 'BACKGROUND'):`

Thank you so much @saifurrehman4114

Fatemehshr, 2021-06-28:
> 
> 
> > https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet.py#L69
> > you can also try this . I am ignoring the background after detection
> > `if (net.GetClassDesc(detection.Instance) != 'BACKGROUND'):`
> 
> Thank you so much @saifurrehman4114

Fatemehshr, 2021-06-28:
> 
> 
> https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet.py#L69
> 
> you can also try this . I am ignoring the background after detection
> 
> `if (net.GetClassDesc(detection.Instance) != 'BACKGROUND'):`

Thank you, I tried, but it works wrong!

Fatemehshr, 2021-06-28:
@saifurrehman4114 before print command, I 've written if (net.GetClassDesc(detection.Instance) != 'BACKGROUND'): , Is it correct?

Fatemehshr, 2021-06-28:
> 
> 
> Hi @Fatemehshr, if you change the class names in the model's `labels.txt` that you want to ignore to `void`, it will ignore them during detection.

Thank you so much

dusty-nv, 2021-06-28:
> if (net.GetClassDesc(detection.Instance) != 'BACKGROUND'):

Try changing that to `if (net.GetClassDesc(detection.ClassID) != 'BACKGROUND'):`, that should work then

Fatemehshr, 2021-06-28:
Dear Dustin,
I change the class names in labels.txt  to void. but it doesn't detect
bird.I also try it
 for person ( I ignore the other labels).
I attached labels.txt. Can you check it? Is it correct?
and I have a question about tis code, how far does it detect birds?

void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
bird
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void
void

dusty-nv, 2021-06-29:
Does it detect `bird` in the same image before you made the change?

Fatemehshr, 2021-06-29:
yes, it detected bird. I sent a snap shot from the result.

saifurrehman4114, 2021-06-30:
Do if (net.GetClassDesc(detection.Instance) == 'Bird'):

Else:

	print("not Bird")

It will work as it will only detect birds

Fatemehshr, 2021-06-30:
> 
> 
> OK, is it the `ssd_coco_labels.txt` that you changed? For that SSD-Mobilenet-v2 COCO model, the labels file is actually called `ssd_coco_labels.txt`
> 
> Have you tried lowering the detection threshold with the `--threshold` argument?

@dusty-nv, Yes, as you said I changed 'ssd_coco_labels.txt ' so it works correctly. Thank you for your helping. Also, I have a question about adjusting or changing object(or bird) detection distance to the camera. It recognizes the bird at a short distance. Is it correct for changing 'threshold' in ' my-detection.py' ?
**net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)**

dusty-nv, 2021-06-30:
If you aren't getting detections, then yes lowering the threshold may help (although it can also introduce false positives)

Fatemehshr, 2021-06-30:
> 
> 
> If you aren't getting detections, then yes lowering the threshold may help (although it can also introduce false positives)

Ok. I have used ' IMX219 Camera, 800 megapixels and 160 FOV'. Do you think this camera is suitable for detection at a bit long distance?

dusty-nv, 2021-06-30:
I'm not sure, best thing would be to test it

Fatemehshr, 2021-08-11:
Hi @dusty-nv
As I said that I want to detect only birds in real-time. So I changed the class names in the model's labels.txt to void except Bird. It works correctly. Now, I want to attach (to have multiple CSI cameras) another camera. Is it correct to change the camera number to 1 in " camera1 = jetson.utils.videosource("csi://1") "
or what changes do I have to do to other files? I have been changing in  " my-detection.py" , for instance:

import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.3)

camera = jetson.utils.videoSource("csi://0")      
camera1 = jetson.utils.videoSource("csi://1") 

display = jetson.utils.videoOutput("display://0") 
display1 = jetson.utils.videoOutput("display://1")

while display.IsStreaming():
	img = camera.Capture()
	img1 = camera1.Capture()

	detections = net.Detect(img)
	detections = net.Detect(img1)

	display.Render(img)
	display1.Render(img1)

	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
	display1.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))

dusty-nv, 2021-08-11:
That is correct with the camera part, however you would need two displays open for that to work with the two displays.  I think you keep one display, and compost the two `img`/`img1` into one image:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#overlay

segnet.py also shows an example of combining two images into one image for rendering to one display output.

Fatemehshr, 2021-08-11:
@dusty-nv I don't need to display them. My aim is to detect birds in two views of the area. Then, Ring a bell as an alarm for avoiding the birds. So, Is it better to not show them?

dusty-nv, 2021-08-12:
You can totally skip the display if you want, or change it to RTP outputs (and keep the two videoOutput objects like you had it)   then you can remotely view the video if you want.

Fatemehshr, 2021-08-12:
thank you for your time @dusty-nv
Why do you think It's better to keep one display, and compost the two img/img1 into one image? Actually, Can I change " my-detection.py"  like this code below for using 2 cameras (using two networks of the same kind and feeding them separately from each cam instead of just one) on jetson nano?

import jetson.inference
import jetson.utils

net1 = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
net2 = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera1 = jetson.utils.videoSource("csi://0")      # '/dev/video0' for V4L2
camera2 = jetson.utils.videoSource("csi://1")      # '/dev/video0' for V4L2
display1 = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file
display2 = jetson.utils.videoOutput("display://1") # 'my_video.mp4' for file

while display.IsStreaming():
	img1 = camera1.Capture()
	detections1 = net1.Detect(img1)
	display1.Render(img1)
	display1.SetStatus("Object Detection | Network {:.0f} FPS".format(net1.GetNetworkFPS()))

	img2 = camera2.Capture()
	detections2 = net2.Detect(img2)
	display2.Render(img2)
	display2.SetStatus("Object Detection | Network {:.0f} FPS".format(net2.GetNetworkFPS()))

dusty-nv, 2021-08-12:
There is no reason to have two separate network objects (unless they are different models) - you are just loading the network twice.  detectNet is independent of the video source and doesn't keep a state between Detect() calls.  You could feed 100 cameras into the same detectNet object if you wanted to.

I recommend not to create two displays as most people only have one monitor attached to their Jetson, and I haven't played around with multi-monitor setups.

Fatemehshr, 2021-08-12:
Ok, that's right. so how do I change it? As you said, It's correct until camera part like this:

import jetson.inference
import jetson.utils
net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.3)
camera = jetson.utils.videoSource("csi://0")
camera1 = jetson.utils.videoSource("csi://1")

how do I keep one display?

Fatemehshr, 2021-08-12:
> That is correct with the camera part, however you would need two displays open for that to work with the two displays. I think you keep one display, and compost the two `img`/`img1` into one [...]
As you said, I checked segnet.py. But I couldn't completely understand. For instance, 

	if buffers.composite:
		jetson.utils.cudaOverlay(buffers.overlay, buffers.composite, 0, 0)
		jetson.utils.cudaOverlay(buffers.mask, buffers.composite, buffers.overlay.width, 0)

dusty-nv, 2021-08-13:
That code basically composts two images into one bigger image.  The second image is pasted to the right of the first image.

closed 2023-04-27 by dusty-nv