dusty-nv/jetson-inference #1130: A Correct "SSD Mobile Net V2" Training

Grungeby52, 2021-06-28:
Hi,
I want to do a correct tutorial for ssd mobilenet v2. Would shuffle the TrainData set give a better result? Finally, should the names in the test.txt, train.txt, trainval.txt and val.txt files be the same in all?

Best Regards,

Onur Güzeldemirci

dusty-nv, 2021-06-28:
Hi @Grungeby52, the training data is automatically shuffled/randomized by the PyTorch dataloader during training.

Ideally, test.txt train.txt and val.txt would have different files.   trainval.txt is the combination of train.txt and val.txt

Grungeby52, 2021-06-28:
Okay, I got it. Thank you quick answer...
Best Regards

Grungeby52, 2021-07-02:
Hi again,

I train 1000 epochs with 30 classes and 40.000 data... 
Avg Loss down to 1.5 at 120th epoch but after a while it goes up to 1.7 again at 140th epoch

Will Avg Loss drop more after a while?
How much should Avg Loss be?
How much efficiency should be at least for 30 classes?

Best Regards

dusty-nv, 2021-07-02:
Sometimes the loss can go back up temporarily, but it should come back down again in futher epochs.

It's hard to say what a "good" loss is, because it can be dataset-dependent.

joeysantana3, 2023-04-03:
@dusty-nv Hello. I'm a little confused about the trainval.txt file. In my current dataset, I have 150 train images and 28 val images. `train.txt` contains 150 lines with the filenames (minus the extensions), `val.txt` has 28 lines, and `trainval.txt` is just both combined. 

Here is my training command:
```
!python train_ssd.py \
--data=/project74/project74 \
--model-dir=/project74/project74/model \
--dataset-type=voc \
--batch-size=8 \
--epochs=500 \
--resume='project74/project74/model/mb1-ssd-Epoch-499-Loss-1.74681722621123.pth'
```

This is the output:
```
2023-04-03 05:27:50 - Prepare training datasets.
...
2023-04-03 05:27:51 - Train dataset size: 178
2023-04-03 05:27:51 - Prepare Validation datasets.
...
2023-04-03 05:27:51 - Validation dataset size: 178
```

I would expect the train dataset to be 150 and the validation to be 28. Can you explain what I might be doing wrong?

Thanks!

joeysantana3, 2023-04-03:
> @dusty-nv Hello. I'm a little confused about the trainval.txt file. In my current dataset, I have 150 train images and 28 val images. `train.txt` contains 150 lines with the filenames (minus the [...]

I fond in another issue that the ssd training script uses `trainval.txt` for training images and `test.txt` for validation images.

closed 2023-04-27 by dusty-nv