dusty-nv/jetson-inference #1133: detectnet command in Retraining ssd mobilenet

mabdelbari, 2021-06-29:
Hey @dusty-nv 
First of all, thank you for your wonderful tutorial..

I just wanna know which file is executed when i write the command below:

detectnet --model=models/fruit/ssd-mobilenet.onnx --labels=models/fruit/labels.txt 
          --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
            csi://0

as i wanna make some customization for the output such as resizing the output window, title of the window .. etc

Thanks in advance

dusty-nv, 2021-06-29:
Hi @mabdelbari, that is the C++ detectnet example found under jetson-inference/examples/detectnet/detectnet.cpp

There is also detectnet.py from jetson-inference/python/examples/detectnet.py

mabdelbari, 2021-06-29:
I tried to edit on the detectnet.py from jetson-inference/python/examples/detectnet.py 
but nothing changed although I used the command below 

detectnet.py --model=models/fruit/ssd-mobilenet.onnx --labels=models/fruit/labels.txt
--input-blob=input_0 --output-cvg=scores --output-bbox=boxes
csi://0

dusty-nv, 2021-06-29:
After you make changes to ` jetson-inference/python/examples/detectnet.py`, you need to re-run cmake to copy it into the bin directory:

``` bash
$ cd /path/to/your/jetson-inference
$ cd build
$ cmake ../
```

mabdelbari, 2021-06-30:
I tried to edit on both py and cpp files and after editing, I did execute cmake but still no changes happened

dusty-nv, 2021-06-30:
Are you using the docker container, or not docker container?

mabdelbari, 2021-06-30:
Nope. i am not using docker container

dusty-nv, 2021-06-30:
Hmm ok.  If you run `cd jetson-inference/python/examples` and run `detectnet.py` from there, do you see your modifications taking effect?

mabdelbari, 2021-07-01:
i executed the command `detectnet.py --model=../training/detection/ssd/models/fruit/ssd-mobilenet.onnx --labels=../training/detection/ssd/models/fruit/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes csi://0` inside `jetson-inference/python/examples` but still no changes 

and I tried to add python3 before detectnet.py, I started to see some changes but there's no output window

dusty-nv, 2021-07-01:
> and I tried to add python3 before detectnet.py, I started to see some changes but there's no output window

Perhaps the changes you made changed or precluded the use of the output window?

mabdelbari, 2021-07-01:
Nope I only changed some text of the print statements

dusty-nv, 2021-07-01:
Also, I forgot that maybe you need `sudo make install` also to pick up your changes:

``` bash
$ cd /path/to/your/jetson-inference/build
$ cmake ../
$ make
$ sudo make install
```

You should then be able to run your updated `detectnet.py` from anywhere

mabdelbari, 2021-07-01:
Yeah, it works now :) 
Many thanks for your help, I appreciate it :)

I just have another point, I only see a part of the whole image, so how can i fit it on my screen at once?

dusty-nv, 2021-07-01:
> is there any way to calculate the distance of the object from the camera?

Doing that would require depth camera, or known object size and camera calibration.  https://www.pyimagesearch.com/2015/01/19/find-distance-camera-objectmarker-using-python-opencv/

mabdelbari, 2021-07-02:
I am a little confused.. "depth camera" do you mean dual camera?
and how can I know object size in real-time streaming?

dusty-nv, 2021-07-02:
Yes, either a stereo camera or a RGBD camera like the Realsense

closed 2023-04-27 by dusty-nv