dusty-nv/jetson-inference #1134: custom detectnet model. LPD_net from Nvidia

C-monC, 2021-06-30:
Hi,

I am trying to run the license plate detection model from Nvidias [catalog](https://ngc.nvidia.com/catalog/models/nvidia:tlt_lpdnet). I've converted the .tlt to a tensorrt engine using tlt_converter. I placed the engine and test image in the repo and ran the following code:
`detectnet --model=<path to jetson-inference>/jetson-inference/python/examples/lpr_engine.engine --output-bbox=boxes img.png output.png
`

The resulting error :
```
[TRT]    INVALID_ARGUMENT: Cannot find binding of given name: data
[TRT]    failed to find requested input layer data in network
[TRT]    device GPU, failed to create resources for CUDA engine
[TRT]    failed to load /home/oob1/Documents/jetson-inference/python/examples/lpr_engine.engine
[TRT]    detectNet -- failed to initialize.
detectnet:  failed to load detectNet model
```

I've seen there are a few issues with the same error but my problem is I do not have access to an onnx model. 

From the documentation for LPD_net:

> These models are based on NVIDIA DetectNet_v2 detector with ResNet18 as feature extractor.

Is this a supported model configuration out of the box?

dusty-nv, 2021-06-30:
Hi @C-monC, I don't have support for the TLT models in this repo - recommend that you run them with DeepStream or https://github.com/NVIDIA-AI-IOT/tlt-triton-apps

C-monC, 2021-06-30:
Ah thanks, I'll check it out.

closed 2021-06-30 by C-monC