dusty-nv/jetson-inference #1135: Training with custum data and convert to Pascal VOC11

syliabe, 2021-06-30:
Hello,

Please, I am preparing my own database to do the training I am using CVAT.org,
When I download the converted data to Pascal VOC11, I noticed that the annotation files were missing information and mentioned unknow as shown in the following figure
[image]

any suggestions please
thank you

dusty-nv, 2021-06-30:
Did you export from CVAT in pascal VOC format or did you use some other tool to convert it?

Are there any annotations in any of the xml files?  Did you create bounding boxes in CVAT?

[Screenshot from 2021-06-30 11-52-32]<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F86236508%2F123941006-09f25700-d99a-11eb-8d6e-1b264d62a25e.png&data=04%7C01%7Cdustinf%40nvidia.com%7Cefb58caaaeed4a1643f208d93bad52c8%7C43083d15727340c1b7db39efd9ccc17a%7C0%7C0%7C637606437866233944%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=pNTGT4pLPMSr03NqdiQ45n2bJyOCh37lz5NgRE%2BaSAw%3D&reserved=0>

syliabe, 2021-06-30:
Yes I exported from CVAT to Pascal VOC format, what I did is that I took a set of data I put it on CVAT.org then export in Pascal VOC11, suddenly I had Annotation, ImageSets and JPEGImages files
in Annotations we find 
[image]

[image]

saifurrehman4114, 2021-06-30:
As @dusty-nv have said you didn't draw the Bounding box or some other shape for annotation basically you have not annotated the picture.

Kindly do that and paste the result picture.

Thank you!

closed 2023-04-27 by dusty-nv