dusty-nv/jetson-inference #1137: Segmentation fault, permission denied and unable to open shared objects

erkarim85, 2021-07-01:
Hello

I am having some problems running segnet.py
When I try to run segment by typing ./segnet ... it gives me permission denied. 
So, I run it using $python3 segnet.py ... 
When I run it, it gives me Segmentation fault. 
I tried to print inside the segnet code so that I can understand where is the problem, and it stops at import jetson.inference line. 

I read similar issues discussions, and I tried to call imagenet to test if it is working, it told me error while loading shared libraries: libjetson-utili.so: cannot opens shared object file. 

I tried copying the file to the aarch64/bin folder, and it didn't work either.
Any advice?

Thanks,
Karim

dusty-nv, 2021-07-01:
Hi @erkarim85, did you run `make` and `sudo make install` from the build folder?   Can you run `python3 -c 'import jetson.inference'`?

If you continue having problems, you may want to try using the pre-built jetson-inference docker container.

erkarim85, 2021-07-01:
@dusty-nv Thank you for your reply. 
Yes, when I got this problem, I decided to do another manual installation which didn't help for some reason.
Now, in a new place, I did a third installation and after I did the steps for installation from source, I run the docker sh, then I test it and it is now working without this error. 
Thanks again

closed 2021-07-01 by erkarim85